coolsnowwolf/lede #4659: R619AC不能启动, Waiting for root device /dev/ubiblock0_1

reallyimeric, 2020-05-23:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：不能启动，好像一直在等待根设备出现

### (2) 路由器型号和固件版本

A：P&W R619AC 64M，目前的master分支最新（commit `cd6e66d6e1c03d623212e4b812a6fbd8412c6366`）

### (3) 详细日志

A：https://fars.ee/9da1

reallyimeric, 2020-05-23:
manifest https://fars.ee/XsfD
config.seed https://fars.ee/E7qF

AmadeusGhost, 2020-05-23:
要不你刷回之前的固件，然后升级一下opboot/pbboot，上128M吧，64M现在也没条件测了😂

reallyimeric, 2020-05-23:
@AmadeusGhost 好的。我的uboot可以启动构建出的FIT image，这样启动的系统是否可以用于更新bootloader？

AmadeusGhost, 2020-05-23:
> 我的uboot可以启动构建出的FIT image，这样启动的系统是否可以用于更新bootloader？

这个是之前恩山上志平大佬改的uboot吧，我记得进uboot web刷opboot.img后重启就是opboot了

reallyimeric, 2020-05-23:
> > 我的uboot可以启动构建出的FIT image，这样启动的系统是否可以用于更新bootloader？
> 
> 这个是之前恩山上志平大佬改的uboot吧，我记得进uboot web刷opboot.img后重启就是opboot了

好的，感谢。

closed 2020-05-25 by coolsnowwolf