coolsnowwolf/lede #4663: 大神，比较紧急，wndr3800这两天都编译出错，提示wnr8500什么什么，请看以下日志。

Mr-hsing, 2020-05-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译出错

### (2) 路由器型号和固件版本

A：网件 wndr3800

### (3) 详细日志
    ^~~~~~~~
    mdio_addr
arch/mips/ath79/mach-tl-wdr8500-v1.c:105:60: error: missing braces around initializer [-Werror=missing-braces]
 static struct mdio_board_info TL_WDR8500_v1_mdio0_info[] = {
                                                            ^
arch/mips/ath79/mach-tl-wdr8500-v1.c:108:15:
   .phy_addr = 0,
               {}
arch/mips/ath79/mach-tl-wdr8500-v1.c:105:60: error: missing braces around initializer [-Werror=missing-braces]
 static struct mdio_board_info TL_WDR8500_v1_mdio0_info[] = {
                                                            ^
arch/mips/ath79/mach-tl-wdr8500-v1.c:108:15:
   .phy_addr = 0,
               {}
cc1: all warnings being treated as errors
make[7]: *** [arch/mips/ath79/mach-tl-wdr8500-v1.o] Error 1
scripts/Makefile.build:326: recipe for target 'arch/mips/ath79/mach-tl-wdr8500-v1.o' failed
make[6]: *** [arch/mips/ath79] Error 2
scripts/Makefile.build:585: recipe for target 'arch/mips/ath79' failed
make[5]: *** [arch/mips] Error 2
Makefile:1052: recipe for target 'arch/mips' failed
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.180'
make[4]: *** [/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.180/.image] Error 2
Makefile:24: recipe for target '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.180/.image' failed
make[3]: *** [install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ar71xx'
Makefile:13: recipe for target 'install' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#1.70#0.68#2.31
make[2]: *** [target/linux/install] Error 2
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_install' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
##[error]Process completed with exit code 2.

Mr-hsing, 2020-05-24:
这是日志https://pipelines.actions.githubusercontent.com/Ahyzfnr5zkWXCar8A2zvqlyNbviMogNmY26e8Z6yMEkkUOZP1c/_apis/pipelines/1/runs/1/signedlogcontent/3?urlExpires=2020-05-24T13%3A53%3A47.5131568Z&urlSigningMethod=HMACV1&urlSignature=IsPJPeLzanF%2FAnMB%2BJ1PAYSmXeNaiZQGmRf6LziNPDQ%3D

AmadeusGhost, 2020-05-25:
不应该啊，之前编译的时候是过的，我重新编译看看

Mr-hsing, 2020-05-25:
> 不应该啊，之前编译的时候是过的，我重新编译看看

对，就这两天的事情而已，之前都正常，而且，貌似只有wndr3800出这个问题，其他人好像没人反馈过，我和另一个人的3800都是同样的错误，不加其他源进去，也是一样出错，希望大佬可以看看。

AmadeusGhost, 2020-05-25:
> 我和另一个人的3800都是同样的错误

[image]
emmm，我全新编译过了

AmadeusGhost, 2020-05-25:
> 我和另一个人的3800都是同样的错误

有条件本地编译嘛，照着改下试试看
```
--- a/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
+++ b/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
@@ -9,6 +9,7 @@
  */
 
 #include <linux/gpio.h>
+#include <linux/phy.h>
 #include <linux/platform_device.h>
 #include <linux/proc_fs.h>
 #include <linux/spi/spi_gpio.h>
```

Mr-hsing, 2020-05-25:
> 有条件本地编译嘛，照着改下试试看
>
> ```
> --- a/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
> +++ b/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
> @@ -9,6 +9,7 @@
>   */
>
>  #include [...]

首先很感谢您的回复，但由于自己的计算机相对比较慢，所以一般是本地配置后把.config上传触发编译，新手刚来的，可以具体说一下，怎么改源码这块吗？谢谢。

Mr-hsing, 2020-05-25:
> 有条件本地编译嘛，照着改下试试看
>
> ```
> --- a/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
> +++ b/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
> @@ -9,6 +9,7 @@
>   */
>
>  #include [...]

你好，请问是在#include <linux/gpio.h>下面增加#include <linux/phy.h>这一行是吗？请问有同有办法在github上编译时更改呢，自己的配置太慢了，甚至慢于云编译。

AmadeusGhost, 2020-05-25:
> 你好，请问是在#include <linux/gpio.h>下面增加#include <linux/phy.h>这一行是吗？请问有同有办法在github上编译时更改呢，自己的配置太慢了，甚至慢于云编译。

要不你等我的pr merge了再试试看吧

Mr-hsing, 2020-05-25:
> > 你好，请问是在#include <linux/gpio.h>下面增加#include <linux/phy.h>这一行是吗？请问有同有办法在github上编译时更改呢，自己的配置太慢了，甚至慢于云编译。
> 
> 要不你等我的pr merge了再试试看吧

噢，我刚把lean大神的源码fork过来后，自己改一了一下然后云编译，现在在进行中，等下看还会不会出错。假如添加这句进去是顺利了，希望您可以跟大神报告一下代码问题。感谢您的回复。

Mr-hsing, 2020-05-25:
> > 你好，请问是在#include <linux/gpio.h>下面增加#include <linux/phy.h>这一行是吗？请问有同有办法在github上编译时更改呢，自己的配置太慢了，甚至慢于云编译。
> 
> 要不你等我的pr merge了再试试看吧
我刚试着把lean的源码fork，改了，然后在那个自动编译的脚本里，把地址换成我fork过来的地址。结果是一样的错误信息。以下是刚编译的连接：https://github.com/Mr-hsing/3800test2/runs/704907979?check_suite_focus=true

Mr-hsing, 2020-05-25:
> > 你好，请问是在#include <linux/gpio.h>下面增加#include <linux/phy.h>这一行是吗？请问有同有办法在github上编译时更改呢，自己的配置太慢了，甚至慢于云编译。
> 
> 要不你等我的pr merge了再试试看吧
以下是刚刚编译时的日志，已经修改过那个文件了，看起来错误是一样的。
2020-05-25T04:20:05.4400910Z arch/mips/ath79/mach-tl-wdr8500-v1.c:109:15:
2020-05-25T04:20:05.4401007Z    .phy_addr = 0,
2020-05-25T04:20:05.4401113Z                {}
2020-05-25T04:20:05.4441207Z cc1: all warnings being treated as errors
2020-05-25T04:20:05.4467950Z make[7]: *** [arch/mips/ath79/mach-tl-wdr8500-v1.o] Error 1
2020-05-25T04:20:05.4468351Z scripts/Makefile.build:326: recipe for target 'arch/mips/ath79/mach-tl-wdr8500-v1.o' failed
2020-05-25T04:20:05.4477919Z scripts/Makefile.build:585: recipe for target 'arch/mips/ath79' failed
2020-05-25T04:20:05.4478911Z make[6]: *** [arch/mips/ath79] Error 2
2020-05-25T04:20:05.4482756Z make[5]: *** [arch/mips] Error 2
2020-05-25T04:20:05.4483194Z Makefile:1052: recipe for target 'arch/mips' failed
2020-05-25T04:20:05.4483772Z make[4]: *** [/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.180/.image] Error 2
2020-05-25T04:20:05.4487426Z make[3]: *** [install] Error 2
2020-05-25T04:20:05.4487936Z make[5]: Leaving directory '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.180'
2020-05-25T04:20:05.4494663Z Makefile:24: recipe for target '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.180/.image' failed
2020-05-25T04:20:05.4495560Z make[4]: Leaving directory '/workdir/openwrt/target/linux/ar71xx'
2020-05-25T04:20:05.4496009Z Makefile:13: recipe for target 'install' failed
2020-05-25T04:20:05.4496442Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2020-05-25T04:20:05.4496670Z time: target/linux/install#1.59#0.64#2.16
2020-05-25T04:20:05.4497064Z target/Makefile:23: recipe for target 'target/linux/install' failed
2020-05-25T04:20:05.4514695Z make[2]: *** [target/linux/install] Error 2
2020-05-25T04:20:05.4515121Z make[2]: Leaving directory '/workdir/openwrt'
2020-05-25T04:20:05.4515702Z make[1]: *** [/workdir/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
2020-05-25T04:20:05.4517896Z target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_install' failed
2020-05-25T04:20:05.4523621Z make[1]: Leaving directory '/workdir/openwrt'
2020-05-25T04:20:05.4540434Z make: *** [world] Error 2
2020-05-25T04:20:05.4540918Z /workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-05-25T04:20:05.4713938Z ##[error]Process completed with exit code 2.
2020-05-25T04:20:05.5245820Z ##[group]Run df -hT
2020-05-25T04:20:05.5246028Z [36;1mdf -hT[0m
2020-05-25T04:20:05.5297226Z shell: /bin/bash -e {0}
2020-05-25T04:20:05.5297332Z env:
2020-05-25T04:20:05.5297507Z   REPO_URL: https://github.com/Mr-hsing/lede
2020-05-25T04:20:05.5297634Z   REPO_BRANCH: master
2020-05-25T04:20:05.5297741Z   FEEDS_CONF: feeds.conf.default
2020-05-25T04:20:05.5297850Z   CONFIG_FILE: .config
2020-05-25T04:20:05.5297951Z   DIY_P1_SH: diy-part1.sh
2020-05-25T04:20:05.5298058Z   DIY_P2_SH: diy-part2.sh
2020-05-25T04:20:05.5298162Z   SSH_ACTIONS: false
2020-05-25T04:20:05.5298266Z   UPLOAD_BIN_DIR: false
2020-05-25T04:20:05.5298366Z   UPLOAD_FIRMWARE: true
2020-05-25T04:20:05.5298470Z   UPLOAD_COWTRANSFER: false
2020-05-25T04:20:05.5298575Z   UPLOAD_WETRANSFER: false
2020-05-25T04:20:05.5298680Z   TZ: Asia/Shanghai
2020-05-25T04:20:05.5298771Z ##[endgroup]
2020-05-25T04:20:05.5642133Z Filesystem     Type      Size  Used Avail Use% Mounted on
2020-05-25T04:20:05.5642830Z udev           devtmpfs  3.4G     0  3.4G   0% /dev
2020-05-25T04:20:05.5642982Z tmpfs          tmpfs     693M  952K  692M   1% /run
2020-05-25T04:20:05.5643134Z /dev/sdb1      ext4       84G   39G   45G  47% /
2020-05-25T04:20:05.5643291Z tmpfs          tmpfs     3.4G  8.0K  3.4G   1% /dev/shm
2020-05-25T04:20:05.5643448Z tmpfs          tmpfs     5.0M     0  5.0M   0% /run/lock
2020-05-25T04:20:05.5643589Z tmpfs          tmpfs     3.4G     0  3.4G   0% /sys/fs/cgroup
2020-05-25T04:20:05.5643749Z /dev/loop0     squashfs   94M   94M     0 100% /snap/core/9066
2020-05-25T04:20:05.5643907Z /dev/loop1     squashfs   40M   40M     0 100% /snap/hub/43
2020-05-25T04:20:05.5644299Z /dev/sdb15     vfat      105M  3.6M  101M   4% /boot/efi
2020-05-25T04:20:05.5644438Z /dev/sda1      ext4       14G   41M   13G   1% /mnt
2020-05-25T04:20:05.5675273Z ##[group]Run cd openwrt/bin/targets/*/*
2020-05-25T04:20:05.5675519Z [36;1mcd openwrt/bin/targets/*/*[0m
2020-05-25T04:20:05.5675661Z [36;1mrm -rf packages[0m
2020-05-25T04:20:05.5675788Z [36;1mecho "::set-env name=FIRMWARE::$PWD"[0m
2020-05-25T04:20:05.5675933Z [36;1mecho "::set-output name=status::success"[0m
2020-05-25T04:20:05.5727746Z shell: /bin/bash -e {0}
2020-05-25T04:20:05.5727847Z env:
2020-05-25T04:20:05.5728016Z   REPO_URL: https://github.com/Mr-hsing/lede
2020-05-25T04:20:05.5728143Z   REPO_BRANCH: master
2020-05-25T04:20:05.5728253Z   FEEDS_CONF: feeds.conf.default
2020-05-25T04:20:05.5728344Z   CONFIG_FILE: .config
[... 17 lines cut ...]
2020-05-25T04:20:05.5974603Z   CONFIG_FILE: .config
2020-05-25T04:20:05.5974691Z   DIY_P1_SH: diy-part1.sh
2020-05-25T04:20:05.5974795Z   DIY_P2_SH: diy-part2.sh
2020-05-25T04:20:05.5974901Z   SSH_ACTIONS: false
2020-05-25T04:20:05.5975013Z   UPLOAD_BIN_DIR: false
2020-05-25T04:20:05.5975102Z   UPLOAD_FIRMWARE: true
2020-05-25T04:20:05.5975208Z   UPLOAD_COWTRANSFER: false
[... cut ...]
2020-05-25T04:20:06.9382619Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-05-25T04:20:06.9414477Z http.https://github.com/.extraheader
2020-05-25T04:20:06.9423620Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-05-25T04:20:06.9465198Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-05-25T04:20:06.9813696Z Cleaning up orphan processes

AmadeusGhost, 2020-05-25:
>  我刚试着把lean的源码fork，改了，然后在那个自动编译的脚本里，把地址换成我fork过来的地址。结果是一样的错误信息。以下是刚编译的连接：https://github.com/Mr-hsing/3800test2/runs/704907979?check_suite_focus=true

知道问题了，你用4.9内核就能正常编译了，4.14的话我下课再看看

Mr-hsing, 2020-05-25:
> > 我刚试着把lean的源码fork，改了，然后在那个自动编译的脚本里，把地址换成我fork过来的地址。结果是一样的错误信息。以下是刚编译的连接：https://github.com/Mr-hsing/3800test2/runs/704907979?check_suite_focus=true
> 
> 知道问题了，你用4.9内核就能正常编译了，4.14的话我下课再看看
哦，感谢你的回复，但4.14的内核，应该比较适合多拔吧。大神，再问个问题，默认在未加第三方源进去的时候，更新安装feeds时，出现以下警告，这有影响吗？感谢你的回复。
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cpufreq/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'smartmontools', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'btrfs-progs', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-dockerman/Makefile' has a dependency on 'docker-ce', which does not exist
WARNING: Makefile 'package/lean/luci-app-dockerman/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'bash', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-filetransfer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-flowoffload/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a dependency on 'wget', which does not exist

AmadeusGhost, 2020-05-25:
> 再问个问题，默认在未加第三方源进去的时候，更新安装feeds时，出现以下警告，这有影响吗？感谢你的回复。

不影响，这样改下就能编译过了
```
diff --git a/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c b/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
index b5b0e420552..fe97f2d5899 100644
--- a/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
+++ b/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
@@ -9,6 +9,7 @@
  */
 
 #include <linux/gpio.h>
+#include <linux/phy.h>
 #include <linux/platform_device.h>
 #include <linux/proc_fs.h>
 #include <linux/spi/spi_gpio.h>
@@ -105,7 +106,7 @@ static struct ar8327_platform_data TL_WDR8500_v1_ar8337_data = {
 static struct mdio_board_info TL_WDR8500_v1_mdio0_info[] = {
 	{
 		.bus_id = "ag71xx-mdio.0",
-		.phy_addr = 0,
+		.mdio_addr = 0,
 		.platform_data = &TL_WDR8500_v1_ar8337_data,
 	},
 };
```

Mr-hsing, 2020-05-25:
> 不影响，这样改下就能编译过了
>
> ```
> diff --git a/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c b/target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr8500-v1.c
> index b5b0e420552..fe97f2d5899 100644 [...]

哦，修改完这代码后，就可以用4.14的内核编译了吗？那我还是等你更新吧。

Mr-hsing, 2020-05-25:
@AmadeusGhost 大神，现把这行代码加进去后，4.9的内核通过编译了，4.14能解决吗？另外，这个源码得什么时候能fix呢，修复后麻烦告知一下，谢谢。

AmadeusGhost, 2020-05-25:
> @AmadeusGhost 大神，现把这行代码加进去后，4.9的内核通过编译了，4.14能解决吗？另外，这个源码得什么时候能fix呢，修复后麻烦告知一下，谢谢。

4.9之前就能过，这2行是修4.14的，我发pr了，等merge

Mr-hsing, 2020-05-25:
> > @AmadeusGhost 大神，现把这行代码加进去后，4.9的内核通过编译了，4.14能解决吗？另外，这个源码得什么时候能fix呢，修复后麻烦告知一下，谢谢。
> 
> 4.9之前就能过，这2行是修4.14的，我发pr了，等merge

大神，刚修改这两处，编译通过了，4.14的内核，请问什么时候代码合并

PR #4668 (2020-05-25): ar71xx: wdr8500: fix build on 4.14
fix #4663
closed 2020-05-25 by coolsnowwolf