coolsnowwolf/lede #4665: 竞斗云自动重启，查找内核日志。发现UBIFS error，请问要怎么解决

Sunnice, 2020-05-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：出现UBIFS error

### (2) 路由器型号和固件版本

A：竞斗云

### (3) 详细日志

A：[   80.312831] UBIFS error (ubi0:2 pid 2085): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   80.312901] UBIFS error (ubi0:2 pid 2085): ubifs_write_inode: can't write inode 915, error -30
[   85.363455] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.363512] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 835, error -30
[   85.373004] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.380603] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 921, error -30
[   85.391072] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.398912] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 922, error -30
[   85.408370] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.416756] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 923, error -30
[   85.428299] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.433861] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 924, error -30
[   85.444332] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.451670] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 925, error -30
[   85.462010] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.469531] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 926, error -30
[   85.479617] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.487557] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 927, error -30
[   85.496870] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.504971] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 928, error -30
[   85.514569] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   85.522575] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 929, error -30
[   90.552690] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   90.552743] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 690, error -30
[   90.561294] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   90.569388] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 692, error -30
[   95.592582] UBIFS error (ubi0:2 pid 5): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[   95.592630] UBIFS error (ubi0:2 pid 5): ubifs_write_inode: can't write inode 695, error -30
[  105.682326] UBIFS error (ubi0:2 pid 668): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[  105.682376] UBIFS error (ubi0:2 pid 668): ubifs_write_inode: can't write inode 936, error -30
[  105.691250] UBIFS error (ubi0:2 pid 668): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[  105.699662] UBIFS error (ubi0:2 pid 668): ubifs_write_inode: can't write inode 939, error -30
[  110.712259] UBIFS error (ubi0:2 pid 668): make_reservation: cannot reserve 160 bytes in jhead 1, error -30
[  110.712656] UBIFS error (ubi0:2 pid 668): ubifs_write_inode: can't write inode 940, error -30

AmadeusGhost, 2020-05-24:
oaf去掉看看

coolsnowwolf, 2020-05-25:
还是 OAF 的问题

Sunnice, 2020-05-25:
> oaf去掉看看
好的大佬

Sunnice, 2020-05-25:
> 还是 OAF 的问题

好的，大屌。我已经在编译去oaf的固件了，还有个问题想请教一下大屌，我日志后面出现ubifs error是不是和我固件格式有关，我编译好固件后，是使用bin格式固件在后台升级。后面我重刷固件，是在opboot下选择ubi格式，虽然还是出现了路由器自动重启，但是看日志没有了后面的ubifs error了

closed 2020-05-26 by coolsnowwolf