coolsnowwolf/lede #4673: 无线中继 不能自动跟踪信道和频率

chenxudong2020, 2020-05-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无线中继 不能自动跟踪信道和频率
1.有没有计划添加luci-app-autoap插件
2.默认编译完成后b70固件 软件包发行源地址变为
src/gz openwrt_core https://openwrt.proxy.ustclug.org/snapshots/targets/ramips/mt7621/packages
src/gz openwrt_base https://openwrt.proxy.ustclug.org/snapshots/packages/mipsel_24kc/base
src/gz openwrt_helloworld https://openwrt.proxy.ustclug.org/snapshots/packages/mipsel_24kc/helloworld
src/gz openwrt_luci https://openwrt.proxy.ustclug.org/releases/18.06.8/packages/mipsel_24kc/luci
src/gz openwrt_packages https://openwrt.proxy.ustclug.org/snapshots/packages/mipsel_24kc/packages
src/gz openwrt_routing https://openwrt.proxy.ustclug.org/snapshots/packages/mipsel_24kc/routing
 https://openwrt.proxy.ustclug.org/此网址没法用啊 除了上述helloworld是自添加其他的没改
### (2) 路由器型号和固件版本

A：HC5962

### (3) 详细日志

A：

xxp2277, 2020-05-26:
wndr3800根据[这里](https://www.right.com.cn/forum/thread-1514173-1-1.html)勉强编译过一个luci-app-autoap（安装过程有个小错误），据用户反馈说基本正常使用。希望哪位有能力的接手去除cpu架构的限制并完善下。

lloyd18, 2020-05-26:
固件默认的源地址没用了，昨天发个iss被大雕关了，也不说啥情况。

dysl357085918, 2020-05-28:
> 固件默认的源地址没用了，昨天发个iss被大雕关了，也不说啥情况。

直接换源吧，看我的那个贴

closed 2020-07-04 by coolsnowwolf