coolsnowwolf/lede #4677: ubuntu1804, sudo apt-get install -y build-essential asciido..... 报错

jialan22010, 2020-05-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
Package p7zip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package p7zip-full is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package texinfo is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  install-info:i386 info:i386 texinfo-doc-nonfree install-info info

E: Unable to locate package asciidoc
E: Unable to locate package uglifyjs
E: Package 'p7zip' has no installation candidate
E: Package 'p7zip-full' has no installation candidate
E: Unable to locate package msmtp
E: Package 'texinfo' has no installation candidate
E: Unable to locate package xmlto
E: Unable to locate package upx
E: Unable to locate package antlr3
E: Unable to locate package gperf

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

AmadeusGhost, 2020-05-26:
```
Package p7zip is not available, but is referred to by another package.
```
换个apt镜像源然后`sudo apt-get update`试试

Andylive5518, 2020-05-30:
换中科大的源

Andylive5518, 2020-05-30:
> ```
> Package p7zip is not available, but is referred to by another package.
> ```
> 
> 换个apt镜像源然后`sudo apt-get update`试试

有点像钩鱼的

jialan22010, 2020-05-30:
> ```
> Package p7zip is not available, but is referred to by another package.
> ```
> 
> 换个apt镜像源然后`sudo apt-get update`试试

好的，谢谢您，按您的方法做了，这个问题解决了

closed 2020-05-30 by jialan22010