coolsnowwolf/lede #4691: 最新编译，luci-app-ttyd编译进去了，不显示

Atroc-0625, 2020-05-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新编译，luci-app-ttyd编译进去了，不显示

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：

hxlls, 2020-05-28:
[image]
昨天提交的源码编译的，没问题，自己检查咯

Atroc-0625, 2020-05-28:
> [image]
> 昨天提交的源码编译的，没问题，自己检查咯

> [image]
> 昨天提交的源码编译的，没问题，自己检查咯
OpenWrt R20.5.20 / LuCI Master (git-20.141.34051-f750b5f).  这是我编译的版本

Sakurakouji-Runa, 2020-05-28:
我也遇到了同样的问题

cattypapa, 2020-05-28:
可能是你没启动

closed 2020-05-29 by Atroc-0625