coolsnowwolf/lede #4708: ssr模式选择大陆白名单以后，ddns公网ip获取错误

cattypapa, 2020-06-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：就是ssr开了大陆白名单模式之后，获取到的ddns公网ip，就是梯子的ip了，不是家里宽带的公网Ip，这个问题可以通过什么办法解决嘛？往上搜了一圈没找到有效的答案...

### (2) 路由器型号和固件版本

A：N1

### (3) 详细日志

A：

xxp2277, 2020-06-01:
用于检测的 URL，比如http://checkip.dyndns.com，将dyndns.com加入SSR+访问控制，不走代理域名。

chanwang5, 2020-06-01:
检查你的ddns获取你公网ip地址的url，把url域名加到不走代理的名单。保险起见建议您把这个url的域名放到ping测网站ping一下把得到的全部ip加到不走代理的名单。或者找一个没有国外ip的获取ip的url。
还有一个最简单的如果你的路由器是一级路由器直接WAN口获取到的是公网ip就直接以让ddns获取你的WAN口地址即可无需通过url获取ip。你这个问题不是插件的问题建议你自己关了！

cattypapa, 2020-06-02:
> 检查你的ddns获取你公网ip地址的url，把url域名加到不走代理的名单。保险起见建议您把这个url的域名放到ping测网站ping一下把得到的全部ip加到不走代理的名单。或者找一个没有国外ip的获取ip的url。
> 还有一个最简单的如果你的路由器是一级路由器直接WAN口获取到的是公网ip就直接以让ddns获取你的WAN口地址即可无需通过url获取ip。你这个问题不是插件的问题建议你自己关了！

哦哦好的，谢谢大哥的回答，我研究一下

cattypapa, 2020-06-02:
> 用于检测的 URL，比如checkip.dyndns.com，将dyndns.com加入SSR+访问控制，不走代理域名。

谢谢~

closed 2020-07-04 by coolsnowwolf