coolsnowwolf/lede #4710: 求助！树莓派4 32位lan口速率百兆

qq635938204, 2020-06-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
       树莓派4 2GB v1.1硬件版本，使用最新代码编译生成32位固件，开启后使用iperf3测速，lan口速率为百兆，上游路由器为asus ac1900p（68U），使用7类网线连接。
       树莓派4的网口硬件是千兆的，还需要通过什么设置或者软件去修改网口速率吗？
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
[image]
A：

cattypapa, 2020-06-01:
借楼问一下楼主，树莓派4b，编译应该选哪个东西，才会有wifi呀？

ggllioio, 2020-06-17:
没查到4的网卡是什么芯片。不过一般只要能驱动上，速度都是自动协商的，不需要手动指定。楼主要不先刷个Linux试试看能不能协商千兆？排除下硬件和线缆的问题。

qq635938204, 2020-06-17:
> 没查到4的网卡是什么芯片。不过一般只要能驱动上，速度都是自动协商的，不需要手动指定。楼主要不先刷个Linux试试看能不能协商千兆？排除下硬件和线缆的问题。

更换官方的debian测速是千兆的

qq635938204, 2020-06-22:
> 没查到4的网卡是什么芯片。不过一般只要能驱动上，速度都是自动协商的，不需要手动指定。楼主要不先刷个Linux试试看能不能协商千兆？排除下硬件和线缆的问题。

> > 没查到4的网卡是什么芯片。不过一般只要能驱动上，速度都是自动协商的，不需要手动指定。楼主要不先刷个Linux试试看能不能协商千兆？排除下硬件和线缆的问题。
> 
> 更换官方的debian测速是千兆的

找到问题了，是小米的“6类”线的问题，标明cat6实际只能到百兆；更换自制的七类线重新测速能够达到千兆！垃圾小米

closed 2020-06-22 by qq635938204