coolsnowwolf/lede #4714: 如何修改luci界面中的时间显示格式？

Kevin19941121, 2020-06-01:
如题 非常想知道答案？

lphgor, 2020-06-02:
源码：
https://github.com/coolsnowwolf/luci/blob/f750b5ffa252e61b88606400f3da8219e12247f3/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm#L57

路由器的路径是：`/usr/lib/lua/luci/view/admin_status/index.htm`

修改`os.date()`的参数，默认留空是`os.date('%c')`

|格式符|含义|具体示例|
|:----:|:----:|:----:|
| %a | 一星期中天数的简写 | (Fri) |
|%A|一星期中天数的全称|(Wednesday)|
|%b|月份的简写|(Sep)|
|%B|月份的全称|(May)|
|%c|日期和时间|(09/16/98 23:48:10)|
|%d|一个月中的第几天|(28)[0 - 31]|
|%H|24小时制中的小时数|(18)[00 - 23]|
|%I|12小时制中的小时数|(10)[01 - 12]|
|%j|一年中的第几天|(209)[01 - 366]|
|%M|分钟数|(48)[00 - 59]|
|%m|月份数|(09)[01 - 12]|
|%P|上午或下午|(pm)[am - pm]|
|%S|一分钟之内秒数|(10)[00 - 59]|
|%w|一星期中的第几天|(3)[0 - 6 = 星期天 - 星期六]|
|%W|一年中的第几个星期|(2)0 - 52|
|%x|日期|(09/16/98)|
|%X|时间|(23:48:10)|
|%y|两位数的年份|(16)[00 - 99]|
|%Y|完整的年份|(2016)|
|%%|字符串'%'|(%)|

Kevin19941121, 2020-06-02:
> 源码：
>
> https://github.com/coolsnowwolf/luci/blob/f750b5ffa252e61b88606400f3da8219e12247f3/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm#L57 [...]

真的非常感谢您！如果是原版的openwrt中 修改的路径和这个是一样的吗？我很想知道！

lphgor, 2020-06-02:
> 真的非常感谢您！如果是原版的openwrt中 修改的路径和这个是一样的吗？我很想知道！

原版旧版18.06的应该是一样的。
19.07和master分支，源码在这个文件。
我没有刷过原版，所以不知道路由器上的路径，你可以自己搜索一下。
https://github.com/openwrt/luci/blob/068449bf15cbf30c4461b2b9d48fe999e9ed642c/modules/luci-mod-status/htdocs/luci-static/resources/view/status/include/10_system.js#L43-L50

Kevin19941121, 2020-06-02:
> > 真的非常感谢您！如果是原版的openwrt中 修改的路径和这个是一样的吗？我很想知道！
> 
> 
> 
> 原版旧版18.06的应该是一样的。
> 
> 19.07和master分支，源码在这个文件。
> 
> 我没有刷过原版，所以不知道路由器上的路径，你可以自己搜索一下。
> 
> https://github.com/openwrt/luci/blob/068449bf15cbf30c4461b2b9d48fe999e9ed642c/modules/luci-mod-status/htdocs/luci-static/resources/view/status/include/10_system.js#L43-L50

非常感谢您热心的回答 真的非常感谢！如果要搜索的话 是要搜索哪个关键字比较好呢？

Kevin19941121, 2020-06-02:
> 非常感谢您热心的回答 真的非常感谢！如果要搜索的话 是要搜索哪个关键字比较好呢？

我自己努力找了一下 没有找到 明天继续！

lphgor, 2020-06-02:
> 我自己努力找了一下 没有找到 明天继续！

应该是这个文件吧，按照js的语法修改
`/www/luci-static/resources/view/status/include/10_system.js`

Kevin19941121, 2020-06-03:
好的 我回去试试！

Kevin19941121, 2020-06-03:
> > 我自己努力找了一下 没有找到 明天继续！
> 
> 
> 
> 应该是这个文件吧，按照js的语法修改
> 
> `/www/luci-static/resources/view/status/include/10_system.js`

非常感谢！

Kevin19941121, 2020-06-09:
> > 我自己努力找了一下 没有找到 明天继续！
> 
> 
> 
> 应该是这个文件吧，按照js的语法修改
> 
> `/www/luci-static/resources/view/status/include/10_system.js`

最近看了很多资料和相关书籍 一直很好奇您是怎么知道源码对应的路由器路径的 我非常想学习 如果有学习资料我希望可以推荐给我 我想学习一下

closed 2020-07-04 by coolsnowwolf