coolsnowwolf/lede #4718: 把L大的docker移植到了官方op上，报这个错误，希望大哥给看一下怎么解决

cattypapa, 2020-06-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：属实是没辙了，谷歌搜了一大圈没我这个问题的解答...就是把L大的docker插件安装到了官方op19.07.3系统以后，docker已经安装成功，但是pull以后报错，不知道应该怎么设置一下，希望大家给个建议，在此谢过

### (2) 路由器型号和固件版本

A：N1

### (3) 详细日志

A：root@OpenWrt:/tmp/run# docker run -d -p 9000:9000 --name=portainer --restart=alw
ays -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/data portain
er/portainer
Unable to find image 'portainer/portainer:latest' locally
latest: Pulling from portainer/portainer
d1e017099d17: Pull complete
956f65650e6c: Pull complete
Digest: sha256:55c7614b1ad61eabc27214299c42d41bb49e5ef78238c0e5783031f041499284
Status: Downloaded newer image for portainer/portainer:latest
docker: Error response from daemon: error creating aufs mount to /opt/docker/aufs/mnt/fd2497c2032b703d3529e1fc6995710071bd70fe5d7e4e01d871e73145681070-init: mount target=/opt/docker/aufs/mnt/fd2497c2032b703d3529e1fc6995710071bd70fe5d7e4e01d871e73145681070-init data=br:/opt/docker/aufs/diff/fd2497c2032b703d3529e1fc6995710071bd70fe5d7e4e01d871e73145681070-init=rw:/opt/docker/aufs/diff/b2818fa700049b76b1fd2711794b533fec9deaf4b650a95002732ffbc4c38a73=ro+wh:/opt/docker/aufs/diff/7b4639af5b0d97f95207cb4869e11191e974bfaf174ad7738d4c120c42c5ed74=ro+wh,dio,xino=/dev/shm/aufs.xino: invalid argument.
See 'docker run --help'.

garypang13, 2020-06-02:
原版在这里 https://github.com/lisaac/luci-app-dockerman 为啥还要移植呢?

cattypapa, 2020-06-02:
> 原版在这里 https://github.com/lisaac/luci-app-dockerman 为啥还要移植呢?

感觉L大的docker更好一点，可以直接安装portainer，就想折腾一下，但是docker全部都安装好了，pull也能pull成功,但是就是报上面的错误。希望得到一下解答，谢谢拉~

closed 2020-07-04 by coolsnowwolf