coolsnowwolf/lede #4727: frpc未完全汉化问题

SterlingCarter, 2020-06-03:
在中文情况下，显示Frpc运行状态的文字没有汉化。依旧显示Frcp status和The Frp service is not running及The Frp service is running.希望大神能够将这里进行汉化，谢谢！！
[image]

ziyangdaoren, 2020-06-07:
翻译更新了，但界面还是英文的

hyird, 2020-06-08:
> 翻译更新了，但界面还是英文的

翻译分离了  你得重新生成config  编译 luci-i18n-frpc-zh-cn

SterlingCarter, 2020-06-09:
> > 翻译更新了，但界面还是英文的
> 
> 翻译分离了你得重新生成config编译luci-i18n-frpc-zh-cn

已经重新编译固件，感谢

ziyangdaoren, 2020-06-10:
> > 翻译更新了，但界面还是英文的
> 
> 翻译分离了 你得重新生成config 编译 luci-i18n-frpc-zh-cn

重新生成的.config中还是没有luci-i18n-frpc-zh-cn，得手动添加CONFIG_PACKAGE_luci-i18n-frpc-zh-cn=y才行

hyird, 2020-06-11:
> > > > > 翻译更新了，但界面还是英文的
> > > > 
> > > > 
> > > > 翻译分离了 你得重新生成config 编译 luci-i18n-frpc-zh-cn
> > > 
> > > 
> > > 重新生成的.config中还是没有luci-i18n-frpc-zh-cn，得手动添加CONFIG_PACKAGE_luci-i18n-frpc-zh-cn=y才行
> > 
> > 
> > 要把.config删了
> 
> .config删了make menuconfig重新生成后还是没有luci-i18n-frpc-zh-cn，在.config文件中手动添加CONFIG_PACKAGE_luci-i18n-frpc-zh-cn=y 后make menuconfig后save了一下，.config文件中的luci-i18n-frpc-zh-cn又没了，不知道是什么情况。
> [.config.txt](https://github.com/coolsnowwolf/lede/files/4759986/default.config.txt)

不会吧 我自己测了没得问题

EnnawYang, 2020-06-11:
[image]
玄学吧，右边是新git clone的，就有

hyird, 2020-06-11:
> [image]
> 玄学吧，右边是新git clone的，就有
rm -rf ./tmp试试

EnnawYang, 2020-06-11:
> > [image]
> > 玄学吧，右边是新git clone的，就有
> rm -rf ./tmp试试

[image]
OK，其实我也只是顺手测试下而已，这个东西我用不到

hyird, 2020-06-11:
> > > [image]
> > > 玄学吧，右边是新git clone的，就有
> > > rm -rf ./tmp试试
> 
> [image]
> OK，其实我也只是顺手测试下而已，这个东西我用不到
强迫症 自己要用 就修了修

ziyangdaoren, 2020-06-11:
> > [image]
> > 玄学吧，右边是新git clone的，就有
> rm -rf ./tmp试试

好了

SterlingCarter, 2020-06-12:
> > 翻译更新了，但界面还是英文的
> 
> 翻译分离了你得重新生成config编译luci-i18n-frpc-zh-cn

实践证明已经完全汉化，且汉化PO包分离，给作者点赞

closed 2020-06-12 by SterlingCarter