coolsnowwolf/lede #4731: 小白请教编译问题

galed-wang, 2020-06-04:
1.编译错误

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

2. 详细叙述

(1) 具体问题

     编译进行到 ” ./scripts/feeds install -a “这一步后，出现提示
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'kmod-fs-exfat', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'ntfs-3g', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/lean/ksmbd-tools/Makefile' has a dependency on 'avahi-daemon', which does not exist
WARNING: Makefile 'package/lean/ksmbd-tools/Makefile' has a build dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/libidn2/Makefile' has a dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cpufreq/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'smartmontools', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'btrfs-progs', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a dependency on 'docker-ce', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'bash', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-filetransfer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-flowoffload/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frps/Makefile' has a dependency on 'wget', which does not exist
[... 97 lines cut ...]
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'libtasn1/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'rpcsvc-proto/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'python3/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'perl/host', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a build dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/tcpping/Makefile' has a dependency on 'libnet-1.2.x', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray-plugin/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

(2) 然后执行”make menuconfig“ ， 进入Luci 菜单下 Applications菜单后，少了很多选项，连Shadowsocks Plus+服务都没有.......

哪位大侠给看看该咋办？谢谢。

wangfeng35, 2020-06-04:
ssr需要自己修改下添加进去 默认是隐藏的，方法参见：
https://mianao.info/2020/05/05/%E7%BC%96%E8%AF%91%E6%9B%B4%E6%96%B0OpenWrt-PassWall%E5%92%8CSSR-plus%E6%8F%92%E4%BB%B6#%E6%9C%AC%E5%9C%B0%E8%87%AA%E7%BC%96%E8%AF%91

galed-wang, 2020-06-04:
> ssr需要自己修改下添加进去 默认是隐藏的，方法参见：
> https://mianao.info/2020/05/05/%E7%BC%96%E8%AF%91%E6%9B%B4%E6%96%B0OpenWrt-PassWall%E5%92%8CSSR-plus%E6%8F%92%E4%BB%B6#%E6%9C%AC%E5%9C%B0%E8%87%AA%E7%BC%96%E8%AF%91

非常感谢！！！

cattypapa, 2020-06-05:
你这是丢包了啊...建议多执行几次update和install，如果还报warning的话，就重新拉取一下源码，正常时不会报这么多warning的

galed-wang, 2020-06-05:
> 你这是丢包了啊...建议多执行几次update和install，如果还报warning的话，就重新拉取一下源码，正常时不会报这么多warning的

我不知道咋回事，我用的自建VPS开全局模式，其它环节都正常，就是这个环节报warning，但看了上面那位wangfeng35 大侠的网站的介绍，出现warning 后，把feed删掉，重新update，今天编译成功了。

Twinzo1, 2020-06-13:
这是因为先安装本地的插件，再安装feeds的插件导致的warning，执行第二次install就会消失

galed-wang, 2020-06-17:
> 这是因为先安装本地的插件，再安装feeds的插件导致的warning，执行第二次install就会消失

Thanks.

closed 2020-07-04 by coolsnowwolf

garcon26, 2023-06-24:
>十分感謝,失敗了好多次,一直不明所以.

daimaCoder, 2024-11-19:
Makefile 'package/public/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist