coolsnowwolf/lede #4735: 建议加入librespeed,可以内网测速

lqhorochi, 2020-06-04:
项目开源地址：https://github.com/adolfintel/speedtest
希望大雕可以考虑加入

KevinMX, 2020-06-05:
内网测速iperf3还不够用吗……

naoki66, 2020-06-05:
librespeed更方便更直观

shikkoku, 2020-06-07:
https://forum.openwrt.org/t/solved-is-it-possible-to-install-librespeed-speed-test-on-openwrt/65931/7 参考这里

ilovesuyy, 2020-06-08:
librespeed太大了吧  我在docker里装了下，有400M.。。。

shikkoku, 2020-06-09:
> librespeed太大了吧 我在docker里装了下，有400M.。。。

https://github.com/saudiqbal/speed-test-openwrt

有精简版的，不过记得装PHP7-cgi

9l, 2020-06-14:
php 更大

Cathgao, 2020-06-15:
你是完全不考虑32M以下flash大小的设备直接塞进去一个NMP环境？

martin0607, 2020-06-17:
> > librespeed太大了吧 我在docker里装了下，有400M.。。。
> 
> https://github.com/saudiqbal/speed-test-openwrt
> 
> 有精简版的，不过记得装PHP7-cgi

PHP7好像不生效。。不知道为啥
[image]

shikkoku, 2020-06-17:
> > > librespeed太大了吧 我在docker里装了下，有400M.。。。
> > 
> > 
> > https://github.com/saudiqbal/speed-test-openwrt
> > 有精简版的，不过记得装PHP7-cgi
> 
> PHP7好像不生效。。不知道为啥
> [image]

你要装PHP7-CGI。

martin0607, 2020-06-17:
> > > > librespeed太大了吧 我在docker里装了下，有400M.。。。
> > > 
> > > 
> > > https://github.com/saudiqbal/speed-test-openwrt
> > > 有精简版的，不过记得装PHP7-cgi
> > 
> > 
> > PHP7好像不生效。。不知道为啥
> > [image]
> 
> 你要装PHP7-CGI。

[image]
CGI是已经装了的，不知道是不是环境变量问题。我看看

closed 2020-07-04 by coolsnowwolf