coolsnowwolf/lede #4736: 最新代码在v2ray编译错误，menuconfig去掉后编译成功

lydasia, 2020-06-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

lydasia, 2020-06-05:
是饮料+的v2ray模块，保留饮料+去掉v2ray模块没问题

tefiszx, 2020-06-05:
最近v2频频被爆出leaks漏斗，上游开发者的更新迭代速度也快。请@lydasia贴出编译出错的log，我们也可以帮你分析问题。

lydasia, 2020-06-05:
> 最近v2频频被爆出leaks漏斗，上游开发者的更新迭代速度也快。请@lydasia贴出编译出错的log，我们也可以帮你分析问题。

是想贴来着，失误terminal output没存下来

panzeryue, 2020-06-05:
我也遇到类似问题了。存了一个出错部分的log截图：
[image]

caicaicai21, 2020-06-06:
我的执行make disclean完全清理后，再重新编译，可以正常编译通过。要先确定好说明里的依赖都已经安装好了。

lydasia, 2020-06-06:
> 我的执行make disclean完全清理后，再重新编译，可以正常编译通过。要先确定好说明里的依赖都已经安装好了。

我都是尽量重新clone的，依赖也有确认，今天又试了一次也成功了，可能更新过了。

closed 2020-06-06 by lydasia