coolsnowwolf/lede #4740: LAN口无法获取ipv6

Cathgao, 2020-06-06:
路由器WAN6能获取到ipv6公网，但LAN口及下级设备无法获取ipv6地址，ULA已经删除。所有直连光猫的设备包括路由器本身都能正常使用ipv6。路由器没有编译安装任何负载均衡。
LAN的IPV6设置无论是混合模式、中继模式，是否勾上通告路由都没有变化。
[image]
[image]

光猫的ipv6设置：
[image]

从路由ping纯v6站点：
[image]

有关ipv6的编译选项：
```
CONFIG_IPV6=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ip6tables-extra=y
CONFIG_PACKAGE_6in4=y
```

### 路由器型号和固件版本

[image]

Cathgao, 2020-06-06:
禁止ipv6解析也已经关掉

mjyhj, 2020-06-06:
LAN的IPV6你打开了没

ptsa, 2020-06-07:
我的wan6 都没ipv6地址 但是使用都没问题 v6可以上网 老版本可以打开 Flow Offloading
新版本 要关闭Flow Offloading 才能v6上网不知道什么情况

Cathgao, 2020-06-07:
> 我的wan6 都没ipv6地址 但是使用都没问题 v6可以上网 老版本可以打开 Flow Offloading
> 新版本 要关闭Flow Offloading 才能v6上网不知道什么情况

你应该是用路由拨号吧，WAN6是没地址的，LAN和WAN会有。我是光猫拨号路由做下级，WAN6有ipv6，LAN没有

ptsa, 2020-06-08:
> > 我的wan6 都没ipv6地址 但是使用都没问题 v6可以上网 老版本可以打开 Flow Offloading
> > 新版本 要关闭Flow Offloading 才能v6上网不知道什么情况
> 
> 你应该是用路由拨号吧，WAN6是没地址的，LAN和WAN会有。我是光猫拨号路由做下级，WAN6有ipv6，LAN没有

的确是光猫拨号的

zhouym233, 2020-06-10:
[image]
一般情况下，需要获取到ipv6-pd才能继续往下一级分发地址。我这边默认情况下可以正常使用。

Cathgao, 2020-06-11:
> [image]
> 一般情况下，需要获取到ipv6-pd才能继续往下一级分发地址。我这边默认情况下可以正常使用。

那请问PD是光猫的问题还是当地运营商的问题？我也留意到在老家正常时候的时候是WAN6是有PD的，到现在这里就没有。一样都是广东移动

Cathgao, 2020-06-24:
联系移动更换光猫型号：中兴F663N后得到ipv6-pd，问题解决

closed 2020-06-24 by Cathgao

yijijianzaos, 2023-06-28:
我也是遇到这种问题了，路由器访问ipv6正常，下面的手机电脑也获取了ipv6地址，但访问不了ipv6网站，求解

PrinSoul, 2024-03-11:
兄弟，解决问题了吗？
我是电信光猫拨号，下接在华硕路由器(联机模式：Passthough)下的设备都能获取到ipv6地址，但都 ping -6 ipw.cn 不通，路由器本身却可以ping通