coolsnowwolf/lede #4742: x86_64 uefi-gpt更新系统导致系统设置被还原为初始化状态

wangkezun, 2020-06-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86_64 uefi-gpt版本，使用openwrt-x86-64-uefi-gpt-squashfs.img更新时，无法保存原有配置，系统更新重启后恢复为初始化状态。

### (2) 路由器型号和固件版本

A：master分支最新代码。冥王峡谷nuc8i7hvk

### (3) 详细日志

A：使用-T命令检查img包，打印日志如下
Image metadata not found
Reading partition table form bootdisk...
cat: write error: Broken pipe
Reading partition table from image...

wangkezun, 2020-06-07:
[config.txt](https://github.com/coolsnowwolf/lede/files/4741343/config.txt)
编译配置文件

9l, 2020-06-14:
先备份配置，更新完恢复配置

cokyrain, 2020-06-16:
> Kernel modules > Filesystems 选上kmod-fs-f2fs试试

cokyrain, 2020-06-16:
另外  保持上一次的分区大小
> Target Images 
 (  ) Kernel partition size (in MB) 
  (   ) Root filesystem partition size (in MB)

wangkezun, 2020-06-16:
> > Kernel modules > Filesystems 选上kmod-fs-f2fs试试

现在是选择了这个的。

wangkezun, 2020-06-16:
> 先备份配置，更新完恢复配置

也是个方法。不过多这么一道操作还是挺麻烦的。

wangkezun, 2020-06-25:
找了很久没找到解决方案。最终还是把x86转成黑苹果用surge做网关了。。这个issue就关掉了

closed 2020-06-25 by wangkezun