coolsnowwolf/lede #4747: 树梅派4b固件编译时无法同时选择luci-app-docker和docker-ce

guitoushayan, 2020-06-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
用lean大固件编译时路由器型号选择树梅派4b后，luci-app-docker自动消失，但是改为x86型号就始终显示。

删除lean大固件中的luci-app-docker后手动添加luci-app-dockerman软件包后，选择上luci-app-dockerman插件不能选择docker-ce，选择了的docker-ce后luci-app-dockerman整行自动消失，不显示了，只剩下docker-ce和ttyd这两行，不选择docker-ce后luci-app-dockerman自己又出现了。

从openwrt官方package中把docker-ce添加到utils中后，手动在.config中添加CONFIG_PACKAGE_luci-app-dockerman=y后编译

出现如下错误
echo "luci-lib-docker" >> /home/ty/lede/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/luci-lib-docker.default.install
make[3]: Leaving directory '/home/ty/lede/package/luci-lib-docker'
time: package/luci-lib-docker/compile#0.18#0.06#0.25
make[3]: Entering directory '/home/ty/lede/package/utils/docker-ce'
grep: ../containerd/Makefile: No such file or directory
OK: Expected ../containerd/Makefile source version '7ad184331fa3e55e52b890ea95e65ba581ae3429', found ''
grep: ../libnetwork/Makefile: No such file or directory
OK: Expected ../libnetwork/Makefile source version '0941c3f409260d5f05cfa6fc68420d8ad45ee483', found ''
grep: ../runc/Makefile: No such file or directory
OK: Expected ../runc/Makefile source version 'dc9208a3303feef5b3839f4323d9beb36df0a9dd', found ''
grep: ../tini/Makefile: No such file or directory
OK: Expected ../tini/Makefile source version 'fec3683b971d9c3ef73f284f176672c44b448662', found ''
Makefile:36: ../../lang/golang/golang-package.mk: No such file or directory
make[3]: *** No rule to make target '../../lang/golang/golang-package.mk'.  Stop.
make[3]: Leaving directory '/home/ty/lede/package/utils/docker-ce'
time: package/utils/docker-ce/compile#0.08#0.02#0.12
make[2]: *** [package/Makefile:113: package/utils/docker-ce/compile] Error 2
make[2]: Leaving directory '/home/ty/lede'
make[1]: *** [package/Makefile:107: /home/ty/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ty/lede'
make: *** [/home/ty/lede/include/toplevel.mk:227：world] 错误 2

### (1) 具体问题

A：树梅派4b固件 无法添加luci-app-docker或luci-app-dockerman插件

### (2) 路由器型号和固件版本

A：树梅派4b 最新拉取的源码

### (3) 详细日志

A：echo "luci-lib-docker" >> /home/ty/lede/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/luci-lib-docker.default.install
make[3]: Leaving directory '/home/ty/lede/package/luci-lib-docker'
time: package/luci-lib-docker/compile#0.18#0.06#0.25
make[3]: Entering directory '/home/ty/lede/package/utils/docker-ce'
grep: ../containerd/Makefile: No such file or directory
OK: Expected ../containerd/Makefile source version '7ad184331fa3e55e52b890ea95e65ba581ae3429', found ''
grep: ../libnetwork/Makefile: No such file or directory
OK: Expected ../libnetwork/Makefile source version '0941c3f409260d5f05cfa6fc68420d8ad45ee483', found ''
grep: ../runc/Makefile: No such file or directory
OK: Expected ../runc/Makefile source version 'dc9208a3303feef5b3839f4323d9beb36df0a9dd', found ''
grep: ../tini/Makefile: No such file or directory
OK: Expected ../tini/Makefile source version 'fec3683b971d9c3ef73f284f176672c44b448662', found ''
Makefile:36: ../../lang/golang/golang-package.mk: No such file or directory
make[3]: *** No rule to make target '../../lang/golang/golang-package.mk'.  Stop.
make[3]: Leaving directory '/home/ty/lede/package/utils/docker-ce'
time: package/utils/docker-ce/compile#0.08#0.02#0.12
make[2]: *** [package/Makefile:113: package/utils/docker-ce/compile] Error 2
make[2]: Leaving directory '/home/ty/lede'
make[1]: *** [package/Makefile:107: /home/ty/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ty/lede'
make: *** [/home/ty/lede/include/toplevel.mk:227：world] 错误 2

ilovesuyy, 2020-06-08:
好像是默认分区160M太小的原因，我把分区改到500M后luci-app-docker的选项就出现了，你可以试试

guitoushayan, 2020-06-08:
> 好像是默认分区160M太小的原因，我把分区改到500M后luci-app-docker的选项就出现了，你可以试试

难不成是我默认分区太大了？我设置的是5000 /笑哭

aizhouyi, 2020-06-08:
哈哈，这个我在别人的issue看到过这个问题，我搬运一下
[docker编译](https://github.com/SuLingGG/OpenWrt-Rpi/issues/10)

guitoushayan, 2020-06-08:
> 哈哈，这个我在别人的issue看到过这个问题，我搬运一下
> [docker编译](https://github.com/SuLingGG/OpenWrt-Rpi/issues/10)

hahahhahahahha,大佬没看到这两个issue都是我嘛？那个上回是集成上了，但是也不是那么顺利的，稀里糊涂的，我后来重新git clone lean大源码之后，那个法子就死活不能用了

但是，但是今天下午参考了这个之后

> 好像是默认分区160M太小的原因，我把分区改到500M后luci-app-docker的选项就出现了，你可以试试

我又重新git了一份lean大源码，分区改成500后，添加luci-app-udp2raw后发现还是不行，但是去掉luci-app-udp2raw后发现，可以了！！！！现在我怀疑可能这俩包冲突，不能同时编译

guitoushayan, 2020-06-09:
最新成果  我先编译完之后带有dockerman的固件之后，不make clean，在添加udp2raw到packag文件夹中，在编译

(o゜▽゜)o☆[BINGO!]  完成，他俩同时在一个固件中了，很迷 /笑哭

一会儿我刷上固件看看最后确定下

aizhouyi, 2020-06-09:
好吧，我没注意。。不过我手动实践了一下，我怎么没重现你的问题啊。。。。
[image]
[image]
你是不是没有在代理下更新feeds啊？？？玄学。。
SuLing大佬的方案是下载dockerman，然后是makefile大法，然后把lean加入的dockerman删掉。这个我也试了，也成功了啊，
[image]
都是dockerman和docker非要命名成一样的锅，，，你再试试，直接用lean大神的源码，（ps：面板有时候有些配置无法使用，只能命令行用docker，习惯docke命令行操作，面板真的很不完善）我这试了两次都有luci-app-docker，面板后续再自己装portainer就是了。
玄学，莫不是型号选错？BCM27xx，BCM2711 boards（64bit），再看看。

aizhouyi, 2020-06-09:
也都可以选，
[image]
我去翻了commit，这个重命名太骚了，https://github.com/coolsnowwolf/lede/commit/297a116933a3e51e305fa5a3943a663b78b26455

guitoushayan, 2020-06-09:
> 也都可以选，
> [image]
> 我去翻了commit，这个重命名太骚了，[297a116](https://github.com/coolsnowwolf/lede/commit/297a116933a3e51e305fa5a3943a663b78b26455)

可能我没有描述清楚😂，你重新git份lean大源码，把lisaac大佬的luci-app-dockerman，luci-lib-docker和sensec大佬的openwrt-udp2raw，luci-app-udp2raw这4个都扔进去，完了在编译试试看

9l, 2020-06-10:
lisaac 大佬的源文件没有放在 luci-app-dockerman 的子目录里，不能添加到 feeds.conf.default 来自动下载和安装

你可以先试试在 feeds.conf.default 中加入这行
`src-git dockerman https://github.com/9l/luci-app-dockerman`

closed 2020-07-04 by coolsnowwolf