coolsnowwolf/lede #4748: 可以整合一下trojan-go吗?

lqhorochi, 2020-06-08:
可以整合一下trojan-go吗?
源码:https://github.com/p4gefau1t/trojan-go

closed 2020-06-11 by coolsnowwolf

Ai1954, 2020-09-14:
我也希望能整合下trojan-go

norikohxb, 2020-10-02:
我也顶一下，看了youtube上面内网封闭评测。说trojan go比trojan速度更快。系统资源占用更少

Biercenter, 2020-10-03:
> 我也顶一下，看了youtube上面内网封闭评测。说trojan go比trojan速度更快。系统资源占用更少

刚才我也看了腹语哥的视频，确实心动

norikohxb, 2020-10-04:
我也是看了腹语哥的，我很早之前就告诉他要测试对比最好是构建内网的一个服务器。然后他通过这种方式，用假的内网上网方式来测试各种比如trojan、v2ray的速度，这次他用内网了，因为谷歌云经常会抽风，你要是测试的时候正好碰上抽风的时候那么效果自然不好。内网测试最准。
等我把服务器架设好了发现ssrplus不支持trojan-go，然后又去看lienol的passwall竟然发现不知道在哪里可以找到。

Biercenter, 2020-10-04:
> 我也是看了腹语哥的，我很早之前就告诉他要测试对比最好是构建内网的一个服务器。然后他通过这种方式，用假的内网上网方式来测试各种比如trojan、v2ray的速度，这次他用内网了，因为谷歌云经常会抽风，你要是测试的时候正好碰上抽风的时候那么效果自然不好。内网测试最准。
> 等我把服务器架设好了发现ssrplus不支持trojan-go，然后又去看lienol的passwall竟然发现不知道在哪里可以找到。

我是编译固件的时候添加passwall源码，这样固件出来就自带，但是我是没集成trojan-go的，正在重新编译带trojan-go的固件。不过看来下就算ssr plus也可以在本地运行一个trojan-go的客户端，默认开着1080的端口，然后在ssr plus上添加socks代理，应该也是可以的。

Ai1954, 2020-10-20:
SSR Plus 用起来比passwall好用多了，我也在等trojan-go，希望大佬整合下吧。