coolsnowwolf/lede #4750: mac连接IPSec VPN路由器重启严重啊, 望解决下

xiaoweb, 2020-06-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
在公司电信网络,使用mac book 连接家里联通网络IPSec VPN, 10秒内必重启
在手机ios连接情况下30秒内重启.
mac连接IPSec VPN路由器重启严重啊, 望解决下

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
j1900 最新版本
A：

### (3) 详细日志
Tue Jun  9 10:12:23 2020 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Tue Jun  9 10:12:23 2020 authpriv.info ipsec_starter[2430]: Starting strongSwan 5.6.3 IPsec [starter]...
Tue Jun  9 10:12:23 2020 daemon.err modprobe: ah4 is already loaded
Tue Jun  9 10:12:23 2020 daemon.err modprobe: esp4 is already loaded
Tue Jun  9 10:12:23 2020 daemon.err modprobe: ipcomp is already loaded
Tue Jun  9 10:12:23 2020 daemon.err modprobe: xfrm4_tunnel is already loaded
Tue Jun  9 10:12:23 2020 daemon.err modprobe: xfrm_user is already loaded
Tue Jun  9 10:12:23 2020 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.6.3, Linux 4.19.122, x86_64)
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG]   loaded IKE secret for %any
Tue Jun  9 10:12:23 2020 daemon.info : 00[CFG]   loaded EAP secret for zhouVpn
Tue Jun  9 10:12:23 2020 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Tue Jun  9 10:12:23 2020 daemon.info : 00[JOB] spawning 16 worker threads
Tue Jun  9 10:12:23 2020 authpriv.info ipsec_starter[2430]: charon (2481) started after 60 ms
A：

xiaoweb, 2020-06-11:
是一走科学上网就重启.  其他正常用好像是好的

RayleiLi, 2020-06-11:
我现在使用softvpn代替ipsec server了

xiaoweb, 2020-06-11:
> 我现在使用softvpn代替ipsec server了

怎么样， 稳定吗

closed 2020-06-12 by xiaoweb