coolsnowwolf/lede #4756: ss时好时坏，反复重启服务

KnowledegAndAction, 2020-06-09:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题
ssr Plus一直显示未运行，ss服务器节点都已添加且可以连通显示ping值，其他设备也可以正常使用此ss节点。设置了socks5 proxy server。

Status里面
Global Client Not Running 
DNS Anti-pollution Running
Global SOCKS5 Proxy Server Not Running 

实际使用情况是浏览网页时比如google，时好时坏，非常不稳定，日志里看到ss一直在重启。

### (2) 路由器型号和固件版本
网件4300， OpenWrt  R20.4.8 / LuCI Master (git-20.088.29068-cebe120)
--
Kernel Version |  4.9.215

### (3) 详细日志
Tue Jun  9 16:24:51 2020 user.notice shadowsocksr: ssr redir tcp error.restart!
Tue Jun  9 16:24:53 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun  9 16:24:53 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun  9 16:24:57 2020 cron.info crond[31873]: crond (busybox 1.30.0) started, log level 8
Tue Jun  9 16:24:58 2020 daemon.info /usr/bin/ss-redir[32054]: initializing ciphers... aes-256-gcm
Tue Jun  9 16:24:58 2020 daemon.info /usr/bin/ss-redir[32054]: listening at 0.0.0.0:1234
Tue Jun  9 16:24:58 2020 daemon.info /usr/bin/ss-redir[32054]: tcp port reuse enabled
Tue Jun  9 16:24:58 2020 daemon.info /usr/bin/ss-redir[32054]: running from root user
Tue Jun  9 16:24:58 2020 daemon.info pdnsd[32089]: pdnsd-1.2.9b-par starting.
Tue Jun  9 16:24:59 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun  9 16:24:59 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun  9 16:25:10 2020 daemon.info /usr/bin/ss-local[32278]: using tcp fast open
Tue Jun  9 16:25:10 2020 daemon.info /usr/bin/ss-local[32278]: initializing ciphers... aes-256-gcm
Tue Jun  9 16:25:10 2020 daemon.info /usr/bin/ss-local[32278]: listening at 0.0.0.0:1081
Tue Jun  9 16:25:10 2020 daemon.info /usr/bin/ss-local[32278]: tcp port reuse enabled
Tue Jun  9 16:25:10 2020 daemon.info /usr/bin/ss-local[32278]: udprelay enabled
Tue Jun  9 16:25:10 2020 daemon.info /usr/bin/ss-local[32278]: udp port reuse enabled
Tue Jun  9 16:25:10 2020 daemon.info /usr/bin/ss-local[32278]: running from root user
Tue Jun  9 16:25:40 2020 user.notice shadowsocksr: ssr redir tcp error.restart!
Tue Jun  9 16:25:41 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun  9 16:25:41 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun  9 16:25:45 2020 cron.info crond[32554]: crond (busybox 1.30.0) started, log level 8
Tue Jun  9 16:25:47 2020 daemon.info /usr/bin/ss-redir[32739]: initializing ciphers... aes-256-gcm
Tue Jun  9 16:25:47 2020 daemon.info /usr/bin/ss-redir[32739]: listening at 0.0.0.0:1234
Tue Jun  9 16:25:47 2020 daemon.info /usr/bin/ss-redir[32739]: tcp port reuse enabled
Tue Jun  9 16:25:47 2020 daemon.info /usr/bin/ss-redir[32739]: running from root user
Tue Jun  9 16:25:47 2020 daemon.info pdnsd[304]: pdnsd-1.2.9b-par starting.
Tue Jun  9 16:25:47 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun  9 16:25:48 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun  9 16:25:54 2020 daemon.info /usr/bin/ss-local[449]: using tcp fast open
Tue Jun  9 16:25:54 2020 daemon.info /usr/bin/ss-local[449]: initializing ciphers... aes-256-gcm
Tue Jun  9 16:25:54 2020 daemon.info /usr/bin/ss-local[449]: listening at 0.0.0.0:1081
Tue Jun  9 16:25:54 2020 daemon.info /usr/bin/ss-local[449]: tcp port reuse enabled
Tue Jun  9 16:25:54 2020 daemon.info /usr/bin/ss-local[449]: udprelay enabled
Tue Jun  9 16:25:54 2020 daemon.info /usr/bin/ss-local[449]: udp port reuse enabled
Tue Jun  9 16:25:54 2020 daemon.info /usr/bin/ss-local[449]: running from root user
Tue Jun  9 16:26:24 2020 user.notice shadowsocksr: ssr redir tcp error.restart!
Tue Jun  9 16:26:26 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun  9 16:26:26 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun  9 16:26:29 2020 cron.info crond[662]: crond (busybox 1.30.0) started, log level 8
Tue Jun  9 16:26:31 2020 daemon.info /usr/bin/ss-redir[845]: initializing ciphers... aes-256-gcm
Tue Jun  9 16:26:31 2020 daemon.info /usr/bin/ss-redir[845]: listening at 0.0.0.0:1234
Tue Jun  9 16:26:31 2020 daemon.info /usr/bin/ss-redir[845]: tcp port reuse enabled
Tue Jun  9 16:26:31 2020 daemon.info /usr/bin/ss-redir[845]: running from root user
Tue Jun  9 16:26:31 2020 daemon.info pdnsd[877]: pdnsd-1.2.9b-par starting.
Tue Jun  9 16:26:31 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun  9 16:26:31 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun  9 16:26:38 2020 daemon.info /usr/bin/ss-local[1019]: using tcp fast open
Tue Jun  9 16:26:38 2020 daemon.info /usr/bin/ss-local[1019]: initializing ciphers... aes-256-gcm
Tue Jun  9 16:26:38 2020 daemon.info /usr/bin/ss-local[1019]: listening at 0.0.0.0:1081
Tue Jun  9 16:26:38 2020 daemon.info /usr/bin/ss-local[1019]: tcp port reuse enabled
Tue Jun  9 16:26:38 2020 daemon.info /usr/bin/ss-local[1019]: udprelay enabled
Tue Jun  9 16:26:38 2020 daemon.info /usr/bin/ss-local[1019]: udp port reuse enabled
Tue Jun  9 16:26:38 2020 daemon.info /usr/bin/ss-local[1019]: running from root user

hcym, 2020-06-09:
怕是小路由性能不够，或者插件互相干扰，一直armv8和x86，只有一个代理插件，广告也是用的ssr内置的，非常正常。

coolsnowwolf, 2020-06-10:
端口冲突或者资源不足

KnowledegAndAction, 2020-06-11:
目前应该只跑了一个SSR Plus，其他东西没跑，网件4300性能不够吗？我记得以前放家里接光猫后面使用似乎正常的，没有这个问题。请问端口冲突如何排查，如果不太麻烦点话，大概思路就行。

jyy5206, 2020-06-12:
换个固件版本？
同4300，R20.5.20 一切正常
甚至比 SS 负荷更大的 V2Ray 也正常。

jyy5206, 2020-06-12:
R20.5.20 固件下载：
https://github.com/jyy5206/OpenWRT-firmware-R20.5.20-for-Netgear-4300V1/tree/master

9l, 2020-06-14:
dns 重绑定保护关了试试

ouweiya, 2020-06-24:
这个问题现在有解决方案吗?

closed 2020-07-04 by coolsnowwolf

drydiy, 2021-07-02:
> 端口冲突或者资源不足

正解，用虚拟机装的，单核CPU，256内存，提示：: ssrplus redir tcp error.restart! 错误。换双核 1G内存就好了。

skyhitnow, 2021-12-21:
I also encountered this problem and finally I found the auto-retrieved node information wrongly set the encryption method to none,which after I changed it to the corresponding option the problem then had been solved. I don't think it's the only reason,but it's definitely one of the reasons for the error:`ssrplus redir tcp error.restart!`