coolsnowwolf/lede #4758: 修改默认登录名后保存设置无响应

lariboo, 2020-06-09:
默认登录名root存在较大安全风险，基于此修改root为其他用户名。修改后，设置均衡负载、定时重启等多个插件均出现保存设置无响应的问题。期待修复。

lariboo, 2020-06-09:
修改了如下位置：

1.修改/etc/passwd
将root:x:0:0:root:/root:/bin/ash修改为username:x:0:0:root:/root:/bin/ash

2.修改/etc/shadow
将root:xxxxxx:0:0:99999:7:::修改为username:xxxxxx:0:0:99999:7:::

4.修改/etc/config/rpcd
将option username 'root' 改成 option username 'username'
将option password '$p$root' 改成 option password '$p$username'

4.修改/usr/lib/lua/luci/controller/admin/index.lua
将page.sysauth = “root” 修改为page.sysauth = “username”

lariboo, 2020-06-09:
1.修改/etc/passwd
将root:x:0:0:root:/root:/bin/ash修改为username:x:0:0:root:/root:/bin/ash

2.修改/etc/shadow
将root:xxxxxx:0:0:99999:7:::修改为username:xxxxxx:0:0:99999:7:::

3.修改/etc/config/rpcd
将option username 'root' 改成 option username 'username'
将option password '$p$root' 改成 option password '$p$username'

4.修改/usr/lib/lua/luci/controller/admin/index.lua
将page.sysauth = “root” 修改为page.sysauth = “username”

lariboo, 2020-06-09:
修改了以下位置——

1.修改/etc/passwd
将root:x:0:0:root:/root:/bin/ash修改为username:x:0:0:root:/root:/bin/ash

2.修改/etc/shadow
将root:xxxxxx:0:0:99999:7:::修改为username:xxxxxx:0:0:99999:7:::

3.修改/etc/config/rpcd
将option username 'root' 改成 option username 'username'
将option password '$p$root' 改成 option password '$p$username'

4.修改/usr/lib/lua/luci/controller/admin/index.lua
将page.sysauth = “root” 修改为page.sysauth = “username”

lariboo, 2020-06-09:
修改了以下位置——

1.修改/etc/passwd
将root❌0:0:root:/root:/bin/ash修改为username❌0:0:root:/root:/bin/ash

2.修改/etc/shadow
将root:xxxxxx:0:0:99999:7:::修改为username:xxxxxx:0:0:99999:7:::

3.修改/etc/config/rpcd
将option username 'root' 改成 option username 'username'
将option password '$p$root' 改成 option password '$p$username'

4.修改/usr/lib/lua/luci/controller/admin/index.lua
将e.sysauth = “root” 修改为e.sysauth = “username”

coolsnowwolf, 2020-06-10:
不支持的，必须用 root

9l, 2020-06-14:
root 用户不能改

closed 2020-07-04 by coolsnowwolf