coolsnowwolf/lede #4769: has a dependency on 'luci', which does not exist

lenngennd, 2020-06-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-clash/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-noddos/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-simple-adblock/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-vpnbypass/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-lib-fs/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-ssl-openssl/Makefile' has a dependency on 'luci', which does not exist

### (2) 路由器型号和固件版本

A：x86

ihuaer, 2020-06-11:
重新下载

lenngennd, 2020-06-11:
把feeds是整个文件删除了  重新./scripts/feeds update -a && ./scripts/feeds install -a

closed 2020-06-12 by coolsnowwolf