coolsnowwolf/lede #4775: 最新提交编译出错 404 not found

HashimotoSakura, 2020-06-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
[*] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译过程出错，从openwrt拉去v2ray-core时出现404 not found

### (2) 路由器型号和固件版本

A：Newifi 3D

### (3) 详细日志

A：
```
Hash of the downloaded file does not match (file: b58b4477020034b67a7649341fae65ea7bdc81851cad47ae7e4a591c0663e4c9, requested: ad8a042a334586f9456eccdbe074d4c3b4dad9b5a39c6ee390a5e631d1c389ae) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/v2ray-core-4.24.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/v2ray-core-4.24.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.

```

HashimotoSakura, 2020-06-11:
补充，从https://codeload.github.com/v2fly/v2ray-core/tar.gz/v4.24.2?/v2ray-core-4.24.2.tar.gz下载core时hash值不匹配导致

killerak19871, 2020-06-11:
我也出现同样问题

closed 2020-06-12 by HashimotoSakura