coolsnowwolf/lede #4777: 编译报错

pmxEr5lZwZ, 2020-06-11:
```
tar: linux-4.14.180/arch/arm64/boot/dts/arm: Directory renamed before its status could be extracted
tar: linux-4.14.180/arch/arm64/boot/dts: Directory renamed before its status could be extracted
tar: linux-4.14.180/arch/arm64/boot: Directory renamed before its status could be extracted
tar: linux-4.14.180/arch/arm64: Directory renamed before its status could be extracted
tar: linux-4.14.180/arch: Directory renamed before its status could be extracted
tar: Exiting with failure status due to previous errors
Makefile:104: recipe for target '/home/user/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/linux-4.14.180/.prepared' failed
make[3]: *** [/home/user/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/linux-4.14.180/.prepared] Error 2
make[3]: Leaving directory '/home/user/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#6.48#2.28#8.35
toolchain/Makefile:98: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory '/home/user/lede'
toolchain/Makefile:96: recipe for target '/home/user/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/user/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
/home/user/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```
编译了很久，以为结束了，去bin目录下看根本没有targets这个目录，看来是编译失败了，请问为什么会失败呢？
用的是docker ubuntu 18.04，已经挂了全局代理

wuxiangzhou2010, 2020-06-11:
刚才修了一个 v2ray hash 不匹配的bug, 可以pull 一下试试

ydhang, 2020-06-11:
报同样的错

wuxiangzhou2010, 2020-06-12:
发一下完整的log？

AP0rPi7uQ9, 2020-06-12:
docker里的ubuntu18.04已经失败无数次了！

我发现在 `./scripts/feeds install -a`  这一步的时候有提示Warning， warning部分如下
```
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist 
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist 
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'kmod-fs-exfat', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'ntfs-3g', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/lean/ksmbd-tools/Makefile' has a dependency on 'avahi-daemon', which does not exist
WARNING: Makefile 'package/lean/ksmbd-tools/Makefile' has a build dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/libidn2/Makefile' has a dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cpufreq/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'smartmontools', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'btrfs-progs', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray-plugin/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist
```

pmxEr5lZwZ, 2020-06-12:
@wuxiangzhou2010 
```
Image Name:   MIPS OpenWrt Linux-4.14.180
Created:      Fri Jun 12 04:18:51 2020
Image Type:   MIPS Linux Kernel Image (lzma compressed)
Data Size:    6832549 Bytes = 6672.41 KiB = 6.52 MiB
Load Address: 80000000
Entry Point:  80000000
mv /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-initramfs-kernel.bin.new /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-initramfs-kernel.bin
cp /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-initramfs-kernel.bin /lede/lede/bin/targets/ramips/mt7620/openwrt-ramips-mt7620-hiwifi_hc5661-initramfs-kernel.bin
rm -f /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin
cp /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/vmlinux /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin
cat /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_hiwifi_hc5661.dtb >> /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin
/lede/lede/staging_dir/host/bin/lzma e /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin -lc1 -lp2 -pb2  /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin.new
mkimage -A mips -O linux -T kernel -C lzma -a 0x80000000 -e 0x80000000 -n 'MIPS OpenWrt Linux-4.14.180' -d /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin.new
Image Name:   MIPS OpenWrt Linux-4.14.180
Created:      Fri Jun 12 04:18:51 2020
Image Type:   MIPS Linux Kernel Image (lzma compressed)
Data Size:    1830822 Bytes = 1787.91 KiB = 1.75 MiB
Load Address: 80000000
Entry Point:  80000000
mv /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin.new /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin
[ -f /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin -a -f /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/root.squashfs ]
dd if=/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hiwifi_hc5661-kernel.bin >> /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin
3575+1 records in
3575+1 records out
1830886 bytes (1.8 MB, 1.7 MiB) copied, 0.0204235 s, 89.6 MB/s
dd if=/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/root.squashfs >> /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin
10101+1 records in
10101+1 records out
5171880 bytes (5.2 MB, 4.9 MiB) copied, 0.0348805 s, 148 MB/s
/lede/lede/staging_dir/host/bin/padjffs2 /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin  4 8 16 64 128 256
padding image to 006ae000
padding image to 006b0000
padding image to 006c0000
echo '{  "metadata_version": "1.0", "supported_devices":["hiwifi,hc5661","hc5661"], "version": { "dist": "OpenWrt", "version": "SNAPSHOT", "revision": "r2582-5715e22b", "target": "ramips/mt7620", "board": "hiwifi_hc5661" } }' | fwtool -I - /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin
[ ! -s "/lede/lede/key-build" -o ! -s "/lede/lede/key-build.ucert" -o ! -s "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin" ] || { cp "/lede/lede/key-build.ucert" "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin.ucert" ; usign -S -m "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin" -s "/lede/lede/key-build" -x "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin.sig" ; ucert -A -c "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin.ucert" -x "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin.sig" ; fwtool -S "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin.ucert" "/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin" ; }
cp /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin /lede/lede/bin/targets/ramips/mt7620/openwrt-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin
rm -f /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5761-initramfs-kernel.bin
cp /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/vmlinux-initramfs /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hiwifi_hc5761-initramfs-kernel.bin
[... cut ...]
time: target/linux/install#348.01#24.02#410.88
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/lede/lede'
target/Makefile:19: recipe for target '/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: *** [/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/lede/lede'
/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

wuxiangzhou2010, 2020-06-12:
```
WARNING: Image file /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-iodata_wn-ac1167gr-squashfs-sysupgrade.bin is too big
```
```
cp: cannot stat '/lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-iodata_wn-ac1167gr-squashfs-sysupgrade.bin': No such file or directory
```

生成的bin文件太大。 去掉一些不用的软件试试 @pmxEr5lZwZ

pmxEr5lZwZ, 2020-06-12:
@wuxiangzhou2010 奇怪了，我都是用的默认的选项呀。。。

Twinzo1, 2020-06-13:
> docker里的ubuntu18.04已经失败无数次了！
>
> 我发现在 `./scripts/feeds install -a` 这一步的时候有提示Warning， warning部分如下
>
> ```
> WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist [...]
feeds.conf顺序问题，执行第二次install应该不会出现这些提示了

wuxiangzhou2010, 2020-06-13:
@pmxEr5lZwZ 你的路由器什么型号？确定一下有没被支持
`make menuconfig` 只选择你路由器型号的设置,不要选择default profile之类的设置.  
有些flash/RAM 较小的路由器可能已经不适用这个固件了， 作者之前在issue中讨论去除一些小flash 设备的支持。

pmxEr5lZwZ, 2020-06-13:
@wuxiangzhou2010 5K路由器，w20型号，mt7620a芯片，大佬能看看这个选DEFAULT的话 ok吗？

wuxiangzhou2010, 2020-06-13:
@pmxEr5lZwZ 这个型号没有被支持吧，  一般来说不是随便选一个配置就行的， 需要自己动手修改代码适配硬件（比如flash/ram. led...）， 可以搜搜openwrt 参考适配的提交记录。

如果感觉麻烦， 还是建议换一个大众一点的已经被支持的路由。

可以参考： https://openwrt.org/toh/start

closed 2020-06-16 by pmxEr5lZwZ