coolsnowwolf/lede #4782: 找到一份螃蟹RTL8111/8168的网卡驱动

martin0607, 2020-06-11:
目前的8111/8168网卡用的是8169的驱动，虽然能用，但是感觉不是专用驱动？找到了适配8111/8168的驱动，自己再从螃蟹官网下载更新了下，编译进系统，可以正常识别使用。L大考虑加进去吗？刚好看到加了8125的驱动了
[image]

martin0607, 2020-06-11:
驱动的地址，直接放package下：https://github.com/BROBIRD/openwrt-r8168

EnnawYang, 2020-06-11:
不如发个PR？

RayleiLi, 2020-06-12:
你好，我想请教一下您如果在编译的时候添加驱动呢？

martin0607, 2020-06-12:
> 你好，我想请教一下您如果在编译的时候添加驱动呢？

直接克隆到package里然后菜单里选上kmod r8168就好了

shikkoku, 2020-06-16:
https://github.com/goldkeyber112/openwrt-r8168/releases

有大佬用SDK编译好了IPK包。安装后把r8169的包卸载重启路由即可使用。

shikkoku, 2020-06-19:
https://github.com/coolsnowwolf/lede/commit/ba8a697b767908a4397e11241b5c394bd29f2d21 已经合并进去了吧？

closed 2020-07-04 by coolsnowwolf

wen-bug, 2023-11-05:
> https://github.com/goldkeyber112/openwrt-r8168/releases
> 
> 有大佬用SDK编译好了IPK包。安装后把r8169的包卸载重启路由即可使用。

大佬我安装这个显示没kmod-8168应该怎么弄

weiyideai520, 2023-12-27:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.188-1-db222399ef90ea6bd3453fe08b323a74) for kmod-r8168
 
 是否可以重新编译个。安装不上了