coolsnowwolf/lede #4786: 发现最近编译总是提示如下错误导致编译停止、失败告终。

SterlingCarter, 2020-06-12:
最近编译固件的时候总是报同样一个错误，真的不知道该怎么解决这个错误。
好像这个文件是在编译过程的时候生成的，并不是L大源码里面自带的文件，以下是错误代码：

Makefiles:37 ***missing separator.   stop.

make[3]: *** [/home/angel/lede/build_dir/target-mipsel_24kc_musl/u-boot-2020.04/ .built] Error 2

make[2]: *** [package/boot/uboot-envtools/compile] Error 2

make[1]: *** [/home/angel/lede/build_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2

make: *** [world] Error 2
[image]

SterlingCarter, 2020-06-15:
怎么没有大神回复呢？

cokyrain, 2020-06-16:
好像出问题的都是ubuntu 20.04

Twinzo1, 2020-06-17:
是不是win10子系统

SterlingCarter, 2020-06-22:
不是win10子系统，用的虚拟机ubuntu 14.6

closed 2020-07-04 by coolsnowwolf