coolsnowwolf/lede #4791: UPNP问题

sandyowen, 2020-06-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
     迅雷提示upnp堵塞，transmission提示端口关闭
### (1) 具体问题
    迅雷提示upnp堵塞，transmission提示端口关闭,把防火墙关闭问题依旧
A：

### (2) 路由器型号和固件版本

A：openwrt x86 R20.5.20

### (3) 详细日志

A：

dolphinpaopao, 2020-06-12:
提供信息太少，没有神仙能够解答，今天下午刚编译x86，迅雷无问题。
作者曾说过个别版本有兼容问题，尝试更换不同软件版本。openwrt这边暂时没有解决办法解决这些繁多的兼容问题，所以现在默认是用的2017年版本。
[image]

ziyangdaoren, 2020-06-12:
upnp确实有问题，5月24号编译的x86固件xbox里NAT显示全通，6月编译的固件xbox网络NAT都变成了中等，upnp出现问题，xbox账号都登不上了，还要挂个梯子才能登账号。

ziyangdaoren, 2020-06-12:
> upnp确实有问题，5月24号编译的x86固件xbox里NAT显示全通，6月编译的固件xbox网络NAT都变成了中等，upnp出现问题，xbox账号都登不上了，还要挂个梯子才能登账号。

**详细日志：**

Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.013976] On node 0 totalpages: 1020772
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.013979]   DMA zone: 64 pages used for memmap
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.013980]   DMA zone: 26 pages reserved
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.013982]   DMA zone: 3998 pages, LIFO batch:0
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.014186]   DMA32 zone: 11792 pages used for memmap
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.014188]   DMA32 zone: 754630 pages, LIFO batch:63
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.055076]   Normal zone: 4096 pages used for memmap
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.055078]   Normal zone: 262144 pages, LIFO batch:63
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.067907] x86/hpet: Will disable the HPET for this platform because it's not reliable
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.067918] Reserving Intel graphics memory at [mem 0xbb000000-0xbeffffff]
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068120] ACPI: PM-Timer IO Port: 0x408
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.068126] ACPI: Local APIC address 0xfee00000
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068137] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068141] ACPI: LAPIC_NMI (acpi_id[0x02] high edge lint[0x1])
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068144] ACPI: LAPIC_NMI (acpi_id[0x03] high edge lint[0x1])
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068147] ACPI: LAPIC_NMI (acpi_id[0x04] high edge lint[0x1])
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068163] IOAPIC[0]: apic_id 1, version 32, address 0xfec00000, GSI 0-86
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068169] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068173] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.068178] ACPI: IRQ0 used by override.
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.068180] ACPI: IRQ9 used by override.
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068183] Using ACPI (MADT) for SMP configuration information
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068188] ACPI: HPET id: 0x8086a201 base: 0xfed00000
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068198] smpboot: Allowing 4 CPUs, 0 hotplug CPUs
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068245] [mem 0xbf000000-0xe00f7fff] available for PCI devices
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068250] Booting paravirtualized kernel on bare hardware
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068256] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Fri Jun 12 23:33:05 2020 kern.notice kernel: [    0.068277] random: get_random_bytes called from start_kernel+0x85/0x54d with crng_init=0
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.068292] setup_percpu: NR_CPUS:8 nr_cpumask_bits:8 nr_cpu_ids:4 nr_node_ids:1
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.069025] percpu: Embedded 41 pages/cpu s128280 r8192 d31464 u524288
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.069042] pcpu-alloc: s128280 r8192 d31464 u524288 alloc=1*2097152
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.069045] pcpu-alloc: [0] 0 1 2 3
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.069088] Built 1 zonelists, mobility grouping on.  Total pages: 1004794
Fri Jun 12 23:33:05 2020 kern.notice kernel: [    0.069094] Kernel command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=F601A0E0-69DF-496D-A608-3199C4460002 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.070320] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.070895] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.089515] Calgary: detecting Calgary via BIOS EBDA area
Fri Jun 12 23:33:05 2020 kern.debug kernel: [    0.089518] Calgary: Unable to locate Rio Grande table in EBDA - bailing!
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.116757] Memory: 3868996K/4083088K available (10248K kernel code, 564K rwdata, 2056K rodata, 1056K init, 2392K bss, 214092K reserved, 0K cma-reserved)
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.116854] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.116940] rcu: Hierarchical RCU implementation.
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.116944] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.116948] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.116952] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.116971] NR_IRQS: 4352, nr_irqs: 1024, preallocated irqs: 16
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.117340] Console: colour dummy device 80x25
Fri Jun 12 23:33:05 2020 kern.info kernel: [    0.117778] console [tty0] enabled
[... cut ...]
Fri Jun 12 23:33:17 2020 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Fri Jun 12 23:33:17 2020 daemon.info procd: - init complete -
Fri Jun 12 23:33:18 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jun 12 23:33:18 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jun 12 23:33:21 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jun 12 23:33:21 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

sandyowen, 2020-06-13:
upnptest测试除了前两项全部为failed，而4月左右的固件就全为passed

DrakeShi, 2020-06-18:
> upnp确实有问题，5月24号编译的x86固件xbox里NAT显示全通，6月编译的固件xbox网络NAT都变成了中等，upnp出现问题，xbox账号都登不上了，还要挂个梯子才能登账号。
> 
> **详细日志：**
> 
> Fri Jun 12 23:33:08 2020 user.notice **upnp** daemon: external interface not found, not starting
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.414698] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.424617] igb 0000:04:00.0 eth3: igb: eth3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.428050] br-lan: port 1(eth0) entered blocking state
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.447625] br-lan: port 1(eth0) entered forwarding state
> Fri Jun 12 23:33:09 2020 daemon.notice netifd: Network device 'eth0' link is up
> Fri Jun 12 23:33:09 2020 daemon.notice netifd: bridge 'br-lan' link is up
> Fri Jun 12 23:33:09 2020 daemon.notice netifd: Interface 'lan' has link connectivity
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.454541] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.514224] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.644475] br-lan: port 2(eth1) entered blocking state
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.652220] br-lan: port 2(eth1) entered forwarding state
> Fri Jun 12 23:33:09 2020 daemon.notice netifd: Network device 'eth1' link is up
> Fri Jun 12 23:33:09 2020 daemon.notice netifd: Network device 'eth3' link is up
> Fri Jun 12 23:33:09 2020 daemon.notice netifd: Interface 'wan' has link connectivity
> Fri Jun 12 23:33:09 2020 daemon.notice netifd: Interface 'wan' is setting up now
> Fri Jun 12 23:33:09 2020 kern.info kernel: [ 14.674875] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
> Fri Jun 12 23:33:09 2020 daemon.err insmod: module is already loaded - slhc
> Fri Jun 12 23:33:09 2020 daemon.err insmod: module is already loaded - ppp_generic
> Fri Jun 12 23:33:09 2020 daemon.err insmod: module is already loaded - pppox
> Fri Jun 12 23:33:09 2020 daemon.err insmod: module is already loaded - pppoe
> Fri Jun 12 23:33:09 2020 daemon.info pppd[4285]: Plugin rp-pppoe.so loaded.
> Fri Jun 12 23:33:09 2020 daemon.info pppd[4285]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
> Fri Jun 12 23:33:09 2020 daemon.notice pppd[4285]: pppd 2.4.8 started by root, uid 0
> Fri Jun 12 23:33:10 2020 daemon.err odhcpd[2810]: Failed to send to ff02::1%lan@br-lan (Address not available)
> Fri Jun 12 23:33:14 2020 daemon.info pppd[4285]: PPP session is 32444
> Fri Jun 12 23:33:14 2020 daemon.warn pppd[4285]: Connected to 00:25:82:01:00:32 via interface eth3
> Fri Jun 12 23:33:14 2020 kern.info kernel: [ 19.872383] pppoe-wan: renamed from ppp0
> Fri Jun 12 23:33:14 2020 daemon.info pppd[4285]: Renamed interface ppp0 to pppoe-wan
> Fri Jun 12 23:33:14 2020 daemon.info pppd[4285]: Using interface pppoe-wan
> Fri Jun 12 23:33:14 2020 daemon.notice pppd[4285]: Connect: pppoe-wan <--> eth3
> Fri Jun 12 23:33:14 2020 daemon.info pppd[4285]: syncppp not active
> Fri Jun 12 23:33:14 2020 daemon.info pppd[4285]: Remote message: 0;User(********@xj.cn) Authenticate OK, Request Accept by xj.cn
> Fri Jun 12 23:33:14 2020 daemon.notice pppd[4285]: PAP authentication succeeded
> Fri Jun 12 23:33:14 2020 daemon.notice pppd[4285]: peer from calling number 00:25:82:01:00:32 authorized
> Fri Jun 12 23:33:14 2020 daemon.notice pppd[4285]: local IP address 172.16.23.197
> Fri Jun 12 23:33:14 2020 daemon.notice pppd[4285]: remote IP address 172.16.0.1
> Fri Jun 12 23:33:14 2020 daemon.notice pppd[4285]: primary DNS address 61.128.114.167
> Fri Jun 12 23:33:14 2020 daemon.notice pppd[4285]: secondary DNS address 61.128.114.134
> Fri Jun 12 23:33:14 2020 daemon.notice netifd: Network device 'pppoe-wan' link is up
> Fri Jun 12 23:33:14 2020 daemon.notice netifd: Interface 'wan' is now up
> Fri Jun 12 23:33:14 2020 user.notice mwan3[4630]: Execute ifup event on interface wan (pppoe-wan)
> Fri Jun 12 23:33:15 2020 user.notice mwan3[4630]: Starting tracker on interface wan (pppoe-wan)
> Fri Jun 12 23:33:15 2020 daemon.err miniupnpd[4980]: could not open lease file: /var/upnp.leases
> Fri Jun 12 23:33:15 2020 daemon.notice miniupnpd[4980]: HTTP listening on port 5000
> Fri Jun 12 23:33:15 2020 daemon.notice miniupnpd[4980]: Listening for NAT-PMP/PCP traffic on port 5351
> Fri Jun 12 23:33:15 2020 user.info mwan3[4630]: connection tracking not flushed on interface wan (pppoe-wan) ifup
> Fri Jun 12 23:33:15 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
> Fri Jun 12 23:33:15 2020 cron.info crond[5084]: crond (busybox 1.30.0) started, log level 8
> Fri Jun 12 23:33:17 2020 kern.warn kernel: [ 22.866430] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
> Fri Jun 12 23:33:17 2020 kern.info kernel: [ 22.875899] NFSD: starting 10-second grace period (net f000008f)
> Fri Jun 12 23:33:17 2020 daemon.notice rpc.statd[5322]: Version 2.3.4 starting
> Fri Jun 12 23:33:17 2020 daemon.warn rpc.statd[5322]: Flags: No-Daemon TI-RPC
> Fri Jun 12 23:33:17 2020 daemon.notice rpc.statd[5322]: Initializing NSM state
> Fri Jun 12 23:33:17 2020 daemon.notice rpc.mountd[5323]: Version 2.3.4 starting
> Fri Jun 12 23:33:17 2020 user.info wsdd2[5394]: starting.
> Fri Jun 12 23:33:17 2020 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
[... cut ...] 20 daemon.info procd: - init complete -
> Fri Jun 12 23:33:18 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
> Fri Jun 12 23:33:18 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
> Fri Jun 12 23:33:21 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
> Fri Jun 12 23:33:21 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

和你同样的问题，4月21日编译的X86固件，近一段时间，UPNP打开情况下，xbox时而开放，时而中等

msoayu56, 2020-06-30:
你们把mwan3卸载就好，我的已经好了

ziyangdaoren, 2020-06-30:
> 你们把mwan3卸载就好，我的已经好了

确实是mwan3导致的，卸载了mwan3后upnp就正常了，但没了mwan3多拨就不能用了，只能二选一了。

msoayu56, 2020-07-01:
> > 你们把mwan3卸载就好，我的已经好了
> 
> 确实是mwan3导致的，卸载了mwan3后upnp就正常了，但没了mwan3多拨就不能用了，只能二选一了。

是的，mwan3好像挺多bug的

closed 2020-07-04 by coolsnowwolf

VergilGao, 2021-08-07:
应该是luci界面的bug，实际上我这边mwan3和upnp同时安装了，luci不显示upnp开放的端口，但是实际上申请到了端口，路由器上对应的端口已经打开了。