coolsnowwolf/lede #4792: mksquashfs出错

BoBppy, 2020-06-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译时mksquashfs出错

### (2) 路由器型号和固件版本

A：万恶的K2P

### (3) 详细日志

A：报错的一段
make[3]: Leaving directory '/home/bobppy/lede/tools/quilt'
time: tools/quilt/compile#0.03#0.01#0.05
cc -L/home/bobppy/lede/staging_dir/host/lib   mksquashfs.o read_fs.o action.o swap.o pseudo.o compressor.o sort.o progressbar.o read_file.o info.o restore.o process_fragments.o caches-queues-lists.o gzip_wrapper.o lzma_xz_wrapper.o xz_wrapper.o lzma_xz_options.o /home/bobppy/lede/staging_dir/host/lib/liblzma.a -lpthread -lm -lz -o mksquashfs
/usr/bin/ld: read_fs.o:(.bss+0x0): multiple definition of `fwriter_buffer'; mksquashfs.o:(.bss+0x400bf0): first defined here
/usr/bin/ld: read_fs.o:(.bss+0x8): multiple definition of `bwriter_buffer'; mksquashfs.o:(.bss+0x400bf8): first defined here
/usr/bin/ld: action.o:(.bss+0x0): multiple definition of `fwriter_buffer'; mksquashfs.o:(.bss+0x400bf0): first defined here
/usr/bin/ld: action.o:(.bss+0x8): multiple definition of `bwriter_buffer'; mksquashfs.o:(.bss+0x400bf8): first defined here
/usr/bin/ld: sort.o:(.bss+0x100000): multiple definition of `fwriter_buffer'; mksquashfs.o:(.bss+0x400bf0): first defined here
/usr/bin/ld: sort.o:(.bss+0x100008): multiple definition of `bwriter_buffer'; mksquashfs.o:(.bss+0x400bf8): first defined here
/usr/bin/ld: info.o:(.bss+0x10): multiple definition of `bwriter_buffer'; mksquashfs.o:(.bss+0x400bf8): first defined here
/usr/bin/ld: info.o:(.bss+0x8): multiple definition of `fwriter_buffer'; mksquashfs.o:(.bss+0x400bf0): first defined here
/usr/bin/ld: restore.o:(.bss+0x0): multiple definition of `fwriter_buffer'; mksquashfs.o:(.bss+0x400bf0): first defined here
/usr/bin/ld: restore.o:(.bss+0x8): multiple definition of `bwriter_buffer'; mksquashfs.o:(.bss+0x400bf8): first defined here
/usr/bin/ld: process_fragments.o:(.bss+0x0): multiple definition of `fwriter_buffer'; mksquashfs.o:(.bss+0x400bf0): first defined here
/usr/bin/ld: process_fragments.o:(.bss+0x8): multiple definition of `bwriter_buffer'; mksquashfs.o:(.bss+0x400bf8): first defined here
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:257: mksquashfs] Error 1
make[4]: Leaving directory '/home/bobppy/lede/build_dir/host/squashfskit-v4.14/squashfs-tools'
make[3]: *** [Makefile:42: /home/bobppy/lede/build_dir/host/squashfskit-v4.14/.built] Error 2
make[3]: Leaving directory '/home/bobppy/lede/tools/squashfskit4'
time: tools/squashfskit4/compile#1.19#0.02#1.22
make[2]: *** [tools/Makefile:163: tools/squashfskit4/compile] Error 2
make[2]: Leaving directory '/home/bobppy/lede'
make[1]: *** [tools/Makefile:159: /home/bobppy/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnyyyyynyyyyyyyyyyyyyyynnynynnyyyynynyyy] Error 2
make[1]: Leaving directory '/home/bobppy/lede'
make: *** [/home/bobppy/lede/include/toplevel.mk:227：world] 错误 2

BoBppy, 2020-06-12:
编译机系统是Manjaro Linux，u是i7-9750h

AmadeusGhost, 2020-06-13:
建议用 Debian 系的编译，比如 Ubuntu

BoBppy, 2020-06-13:
自己排查去了

BoBppy, 2020-06-19:
https://bugs.openwrt.org/index.php?do=details&task_id=3104
因为gcc10导致的，我编译时确实删除了gcc检测代码(makelife中只有4.8-9)
openwrt似乎是修复了 这边似乎没有？

BoBppy, 2020-06-19:
prereq-build.mk可作如下修改
https://gitce.net/deny/openwrt/commit/835d1c68a0

BoBppy, 2020-06-19:
Lionel源码中的commit
https://github.com/Lienol/openwrt/commit/059db419d7dfbf8e27c49b1f6b88e8170d659088

BoBppy, 2020-06-19:
https://github.com/coolsnowwolf/lede/issues/4640

BoBppy, 2020-06-19:
允许的话，可以提pr吗

closed 2020-06-22 by BoBppy