coolsnowwolf/lede #4797: bash: mipsel-openwrt-linux-musl-gcc: command not found

pmxEr5lZwZ, 2020-06-13:
安装了CURL库后，重新运行下载dl库的日志，还是有
1.bash: mipsel-openwrt-linux-musl-gcc: command not found
2.grep: /lede/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.180/.config: No such file or directory

```
user@56cbf16193f9:/lede/lede$ make -j1 download V=s
make[1]: Entering directory '/lede/lede'
make[2]: Entering directory '/lede/lede/tools/gmp'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/gmp'
time: tools/gmp/download#0.00#0.00#0.07
make[2]: Entering directory '/lede/lede/tools/mpfr'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/mpfr'
time: tools/mpfr/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/mpc'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/mpc'
time: tools/mpc/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/libelf'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/libelf'
time: tools/libelf/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/expat'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/expat'
time: tools/expat/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/m4'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/m4'
time: tools/m4/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/libtool'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/libtool'
time: tools/libtool/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/autoconf'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/autoconf'
time: tools/autoconf/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/automake'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/automake'
time: tools/automake/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/flex'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/flex'
time: tools/flex/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/bison'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/bison'
time: tools/bison/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/pkg-config'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/pkg-config'
time: tools/pkg-config/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/mklibs'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/mklibs'
time: tools/mklibs/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/zlib'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/zlib'
time: tools/zlib/download#0.00#0.00#0.07
make[2]: Entering directory '/lede/lede/tools/sstrip'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/sstrip'
time: tools/sstrip/download#0.00#0.00#0.06
make[2]: Entering directory '/lede/lede/tools/make-ext4fs'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/tools/make-ext4fs'
[... 499 lines cut ...]
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/package/utils/jsonfilter'
time: package/utils/jsonfilter/download#0.07#0.03#0.50
make[2]: Entering directory '/lede/lede/package/utils/lua'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/package/utils/lua'
time: package/utils/lua/download#0.08#0.00#0.24
make[2]: Entering directory '/lede/lede/package/utils/mtd-utils'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/package/utils/mtd-utils'
time: package/utils/mtd-utils/download#0.06#0.01#0.18
make[2]: Entering directory '/lede/lede/package/utils/util-linux'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/lede/lede/package/utils/util-linux'
time: package/utils/util-linux/download#0.12#0.01#0.24
make[1]: Leaving directory '/lede/lede'
make[1]: Entering directory '/lede/lede'
make[2]: Entering directory '/lede/lede/target/linux'
make[3]: Entering directory '/lede/lede/target/linux/ramips'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/lede/lede/target/linux/ramips'
make[2]: Leaving directory '/lede/lede/target/linux'
time: target/linux/download#0.00#0.01#0.09
make[1]: Leaving directory '/lede/lede'
```

pmxEr5lZwZ, 2020-06-13:
具体下载信息参照这个issue
#4796

closed 2020-06-16 by pmxEr5lZwZ

sonyxu, 2020-06-27:
> 具体下载信息参照这个issue
> #4796

请问一下，最后有没有解决呢 这是什么导致的，又如何解决能否告知
我先尝试编译一下，能过就上来说一句

CYaNu, 2022-08-21:
更新一下.config就行了。一般都是这个问题。先clean然后make menuconfig，save一下再编译就行了。