coolsnowwolf/lede #4803: 从来没有编译成功过，这次好不容易编译了很久，又报错

pmxEr5lZwZ, 2020-06-13:
按照教程来的，从来没有编译成功过，
（这三天内的经历）
1.之前用docker ubuntu 18.04编译，不行，
2.后来用windows商店Ubuntu 18.04 LTS编译，开全局代理，不行
3.海外VPS用Docker ubuntu 18.04编译，不行，
4.海外VPS用make menuconfig不选型号，默认编译，不行，
5.windows商店Ubuntu 20.04 LTS 编译，选择Newifi D2型号，不行
是我的人品不行么。。。

```
touch /home/user/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.opkg_installed
mkdir -p /home/user/lede/bin/targets/ramips/mt7621/packages /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/CONTROL /home/user/lede/staging_dir/target-mipsel_24kc_musl/pkginfo
install -d -m0755 /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/usr/lib/opkg
install -d -m0755 /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/bin
install -d -m0755 /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/etc/opkg
install -d -m0755 /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/etc/uci-defaults
install -m0644 ./files/customfeeds.conf /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/etc/opkg/customfeeds.conf
install -m0644 ./files/opkg.conf /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/etc/opkg.conf
install -m0755 ./files/20_migrate-feeds /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/etc/uci-defaults/
install -m0755 /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-install/usr/bin/opkg-cl /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/bin/opkg
install -d -m0755 /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/usr/sbin
install -m0755 ./files/opkg-key /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/usr/sbin/
find /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="mipsel-openwrt-linux-musl-"  NO_RENAME=1 ; NM="mipsel-openwrt-linux-musl-nm" STRIP="/home/user/lede/staging_dir/host/bin/sstrip" STRIP_KMOD="/home/user/lede/scripts/strip-kmod.sh" PATCHELF="/home/user/lede/staging_dir/host/bin/patchelf" /home/user/lede/scripts/rstrip.sh /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg
rstrip.sh: /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/bin/opkg: executable
(cd /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -x "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm; echo "$V_Package_opkg_conffiles" > conffiles;  )
install -d -m0755 /home/user/lede/bin/packages/mipsel_24kc/base
/home/user/lede/scripts/ipkg-build -c -o 0 -g 0 /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg /home/user/lede/bin/packages/mipsel_24kc/base
/home/user/lede/staging_dir/host/bin/find: '/home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg/etc/opkg/keys/': No such file or directory
Packaged contents of /home/user/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/opkg-2020-01-25-c09fe209/ipkg-mipsel_24kc/opkg into /home/user/lede/bin/packages/mipsel_24kc/base/opkg_2020-01-25-c09fe209-1_mipsel_24kc.ipk
echo "opkg" >> /home/user/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/opkg.default.install
rm -rf /home/user/lede/tmp/stage-opkg
mkdir -p /home/user/lede/tmp/stage-opkg/host /home/user/lede/staging_dir/target-mipsel_24kc_musl/packages /home/user/lede/staging_dir/host/packages
find /home/user/lede/tmp/stage-opkg -name '*.la' | xargs -r rm -f;      find /home/user/lede/tmp/stage-opkg -name '*.la' | xargs -r rm -f;
if [ -f /home/user/lede/staging_dir/target-mipsel_24kc_musl/packages/opkg.list ]; then /home/user/lede/scripts/clean-package.sh "/home/user/lede/staging_dir/target-mipsel_24kc_musl/packages/opkg.list" "/home/user/lede/staging_dir/target-mipsel_24kc_musl"; fi
if [ -d /home/user/lede/tmp/stage-opkg ]; then (cd /home/user/lede/tmp/stage-opkg; find ./ > /home/user/lede/tmp/stage-opkg.files);     SHELL= flock /home/user/lede/tmp/.staging-dir.flock -c ' mv /home/user/lede/tmp/stage-opkg.files /home/user/lede/staging_dir/target-mipsel_24kc_musl/packages/opkg.list && cp -fpR /home/user/lede/tmp/stage-opkg/* /home/user/lede/staging_dir/target-mipsel_24kc_musl/; '; fi
rm -rf /home/user/lede/tmp/stage-opkg
touch /home/user/lede/staging_dir/target-mipsel_24kc_musl/stamp/.opkg_installed
make[3]: Leaving directory '/home/user/lede/package/system/opkg'
time: package/system/opkg/compile#5.25#7.64#14.53
make[3]: Entering directory '/home/user/lede/package/system/urandom-seed'
touch /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/.prepared_fdd139aac3e84ba8cfac6482cb90c809_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed
touch /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/.prepared_fdd139aac3e84ba8cfac6482cb90c809_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/.configured_*
rm -f /home/user/lede/staging_dir/target-mipsel_24kc_musl/stamp/.urandom-seed_installed
(cd /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/./; if [ -x ./configure ]; then find /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/ -name config.guess | xargs -r chmod u+w; find /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/ -name config.guess | xargs -r -n1 cp --remove-destination /home/user/lede/scripts/config.guess; find /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/ -name config.sub | xargs -r chmod u+w; find /home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed/ -name config.sub | xargs -r -n1 cp --remove-destination /home/user/lede/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed:urandom-seed -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed:urandom-seed -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/user/lede/build_dir/target-mipsel_24kc_musl/urandom-seed:urandom-seed -Wformat -Werror=forma [... cut ...]
make[3]: *** [Makefile:133: /home/user/lede/build_dir/target-mipsel_24kc_musl/busybox-1.30.0/.configured_68b329da9893e34099c7d8ad5cb9c940_68bcc60892ef01f30ea83ed50d912b87] Error 2
make[3]: Leaving directory '/home/user/lede/package/utils/busybox'
time: package/utils/busybox/compile#2.92#5.46#12.61
make[2]: *** [package/Makefile:113: package/utils/busybox/compile] Error 2
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:107: /home/user/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:227: world] Error 2
```

aizhouyi, 2020-06-13:
实体机编译，真的很多莫名奇妙问题，成功率低的可怕，试试github的action吧，用了action我已经彻底放弃实体机编译了

pmxEr5lZwZ, 2020-06-14:
@aizhouyi 非常感谢，不过有个疑问，第一次编译的config没有ssrplus，如果我想第二次注释HELLOWORLD+勾选后，我应该这么做？有没有workflow的模板可以分享一下呢？谢谢

aizhouyi, 2020-06-14:
一一解答
第一，可以通过修改feeds文件实现，确实可以。但我用的不是这种方法。
第二，我是直接下载科学源码，再去自动生成feeds，云action推荐。
https://github.com/P3TERX/Actions-OpenWr
这个写的非常完善了，文档十分详细。
第三，模板很多了，很多定时编译的大佬一直更新，比如，https://github.com/SuLingGG/OpenWrt-Rpi。你也可以点进我的仓库可以看到我的，主要修改是workflow文件（只是我开启了国内上传固件，其他全是默认）和diy-part1.sh（这个就是增加你想要的功能），把本地的配置文件推上去，自动就开始编译了

aizhouyi, 2020-06-14:
> @aizhouyi 非常感谢，不过有个疑问，第一次编译的config没有ssrplus，如果我想第二次注释HELLOWORLD+勾选后，我应该这么做？有没有workflow的模板可以分享一下呢？谢谢

一一解答
第一，可以通过修改feeds文件实现，确实可以。但我用的不是这种方法。
第二，我是直接下载科学源码，再去feeds。
云action推荐
https://github.com/P3TERX/Actions-OpenWr
这个写的非常完善了，文档十分详细。
第三，模板很多了，很多定时编译的大佬一直更新，比如
https://github.com/SuLingGG/OpenWrt-Rpi
你也可以点进我的仓库可以看到我的，
https://github.com/aizhouyi/openwrt_compile.git
主要修改是workflow文件（只是我开启了国内上传固件，其他全是默认）和diy-part1.sh（这个就是增加你想要的功能），把本地的配置文件推上去，自动就开始编译了。

closed 2020-06-14 by pmxEr5lZwZ

Jack-s-cmd, 2020-06-19:
到这里的报错基本就快编译好了，我自己遇到的是冲突造成的，你检查一下，我之前不知道已经集成了wifi驱动，自己又装了一个，最后冲突了。我建议，git下来后，什么都不要改，直接满线程跑一遍代码，一切正常，后面都好解决了，编译错了重新再来，后面再编译我用VPS跑10分钟不到就可以跑完，美滋滋。