coolsnowwolf/lede #4804: mwan3无法正常使用

miudin, 2020-06-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mwan3,配置后一直卡在配置中,直接修改mwan3配置文件，网口无法上线。官方的mwn3是正常的，或者L大1月份的源码也是正常的。

### (2) 路由器型号和固件版本
x86-64
A：

### (3) 详细日志

A：

sillycat3, 2020-06-13:
升级最新mwan3版本就能正常了

jshensh, 2020-06-27:
2.6.18 版本执行 ``mwan3 restart`` 一直会卡在那里，升到 2.8.7 以后 balanced 策略可以正常使用，但指定接口和源 IP 的策略不生效，不知道有没有正常一点的版本

sillycat3, 2020-06-28:
> 2.6.18 版本执行 `mwan3 restart` 一直会卡在那里，升到 2.8.7 以后 balanced 策略可以正常使用，但指定接口和源 IP 的策略不生效，不知道有没有正常一点的版本

我是单线双拨，你说的问题我目前都正常。 或者你试下设置好后重启一下mwan3服务看看
BTW：如果搞多线或者Upnp 我觉得 Lienol的固件比较稳定少问题

jshensh, 2020-06-28:
> > 2.6.18 版本执行 `mwan3 restart` 一直会卡在那里，升到 2.8.7 以后 balanced 策略可以正常使用，但指定接口和源 IP 的策略不生效，不知道有没有正常一点的版本
> 
> 我是单线双拨，你说的问题我目前都正常。 或者你试下设置好后重启一下mwan3服务看看
> BTW：如果搞多线或者Upnp 我觉得 Lienol的固件比较稳定少问题

我找到问题了，策略名称我写的 ``srcip_222_all_rule`` 死活不给我加载，也不给我报错，改短一点就好了

closed 2020-07-04 by coolsnowwolf