coolsnowwolf/lede #4819: 编译报错，请问大神怎么解

SterlingCarter, 2020-06-15:
最近编译固件的时候总是报同样一个错误，真的不知道该怎么解决这个错误。
好像这个makefiles文件是在编译过程的时候自动生成的，并不是L大源码里面自带的文件，编译倒到这里提示第37行，缺少空格符，以下是错误代码：

Makefiles:37 ***missing separator. stop.

make[3]: *** [/home/angel/lede/build_dir/target-mipsel_24kc_musl/u-boot-2020.04/ .built] Error 2

make[2]: *** [package/boot/uboot-envtools/compile] Error 2

make[1]: *** [/home/angel/lede/build_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2

make: *** [world] Error 2
[image]

cokyrain, 2020-06-16:
好像出问题的都是ubuntu 20.04

SterlingCarter, 2020-06-18:
> 好像出问题的都是ubuntu 20.04

我用的ubuntu14.06

webguest, 2020-06-24:
同样的问题，ubuntu1804 server

` mkdir -p $(dirname $dest) ; cp -fpR $file $dest ; done ; echo ; link_contents() { local src=$1 ; local dest=$2 ; local dirs dir base ; if [ -n "$(find $src -mindepth 1 -maxdepth 1 -name '*.go' \! -type d)" ]; then echo "$src is already a Go library" ; return 1 ; fi ; dirs=$(find $src -mindepth 1 -maxdepth 1 -type d) ; for dir in $dirs; do base=$(basename $dir) ; if [ -d $dest/$base ]; then case $dir in */usr/share/gocode/src/github.com/xtaci/kcptun) echo "github.com/xtaci/kcptun is already installed. Please check for circular dependencies." ;; *) link_contents $src/$base $dest/$base ;; esac ; else echo "...${src#/home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src}/$base" ; ln -sf $src/$base $dest/$base ; fi ; done ; } ; if [ "" != 1 ]; then if [ -d /home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src ]; then echo "Symlinking directories from /home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src into /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.go_work/build/src" ; link_contents /home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.go_work/build/src ; else echo "/home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src does not exist, skipping symlinks" ; fi ; else echo "Not building binaries, skipping symlinks" ; fi ; echo ; )
Copying files from /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409 into /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.go_work/build/src/github.com/xtaci/kcptun
client/config.go
client/dial.go
client/main.go
client/signal.go
generic/comp.go
generic/copy.go
generic/snmp.go
go.mod
go.sum
server/config.go
server/main.go
server/signal.go

/home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src does not exist, skipping symlinks

touch /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.built
touch /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.built_check
( cd /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.go_work/build ; export      GOOS=linux GOARCH=mipsle GO386= GOARM= GOMIPS=softfloat GOMIPS64= CGO_ENABLED=1 CC=mipsel-openwrt-linux-musl-gcc CXX=mipsel-openwrt-linux-musl-g++ CGO_CFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409:kcptun-20200409 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/deb/lede/staging_dir/target-mipsel_24kc_musl/include -I/home/deb/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/include -I/home/deb/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include/fortify -I/home/deb/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409:kcptun-20200409 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/deb/lede/staging_dir/target-mipsel_24kc_musl/lib -L/home/deb/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/lib -L/home/deb/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/lib -znow -zrelro"        GOPATH=/home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.go_work/build GOCACHE=/home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.go_work/cache GOENV=off ; echo "Finding targets" ; targets=$(go list github.com/xtaci/kcptun/...) ; for pattern in ; do targets=$(echo "$targets" | grep -v "$pattern") ; done ; echo ; if [ "" = 1 ]; then echo "Calling go generate" ; go generate -v  $targets ; echo ; fi ; if [ "" != 1 ]; then echo "Building targets" ; case mipsle in arm)             installsuffix="v" ;; mips|mipsle)     installsuffix="softfloat" ;; mips64|mips64le) installsuffix="" ;; esac ; ldflags="-linkmode external -extldflags '-L/home/deb/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/deb/lede/staging_dir/target-mipsel_24kc_musl/lib -L/home/deb/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/lib -L/home/deb/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/lib -Wl,-z,now -Wl,-z,relro'" ; pkg_gcflags="" ; pkg_ldflags="-s -w -X 'main.VERSION=20200409-1 for OpenWrt'" ; for def in ; do pkg_ldflags="$pkg_ldflags -X $def" ; done ; go install  ${installsuffix:+-installsuffix $installsuffix} -trimpath -ldflags "all=$ldflags" -v ${pkg_gcflags:+-gcflags "$pkg_gcflags"} ${pkg_ldflags:+-ldflags "$pkg_ldflags $ldflags"}  $targets ; retval=$? ; echo ; if [ "$retval" -eq 0 ] && [ -z "$(find /home/deb/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200409/.go_work/build/bin/linux_mipsle -maxdepth 0 -type d \! -empty 2>/dev/null)" ]; then echo "No binaries were generated, consider adding GO_PKG_SOURCE_ONLY:=1 to Makefile" ; echo ; fi ; echo "Cleaning module download cache (golang/go#27455)" ; go clean -mo [... cut ...]
package/Makefile:111: recipe for target 'package/lean/kcptun/compile' failed
make[2]: *** [package/lean/kcptun/compile] Error 2
make[2]: Leaving directory '/home/deb/lede'
package/Makefile:107: recipe for target '/home/deb/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/deb/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/deb/lede'
/home/deb/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
[image]
`

hzjnet, 2020-06-28:
> 同样的问题，ubuntu1804 server
>
> ` mkdir -p $(dirname $dest) ; cp -fpR $file $dest ; done ; echo ; link_contents() { local src=$1 ; local dest=$2 ; local dirs dir base ; if [ -n "$(find $src -mindepth 1 [...]

其实解决也简单，执行以下2行即可
 export GO111MODULE=on 
 export GOPROXY=https://goproxy.cn,direct

closed 2020-07-04 by coolsnowwolf