coolsnowwolf/lede #4820: X86版在内网通过公网IP无法访问内网设备

zeruns, 2020-06-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在内网里使用公网IP访问无法正常访问内网设备，已做端口映射，但在外网可以正常访问，内网只能使用内网IP访问内网设备

### (2) 路由器型号和固件版本

A：X86软路由，CPU：1037U，网卡RTL8111，固件版本：OpenWrt R20.5.20 / LuCI Master (git-20.141.34051-f750b5f) 

### (3) 相关设置截图

A：[image]
[image]

vlanbin, 2020-06-16:
可以试试从这两个方面找下原因:
1. 启用端口环回
2. 不要编译docker
#3742

zeruns, 2020-06-16:
> 可以试试从这两个方面找下原因:
> 
> 1. 启用端口环回
> 2. 不要编译docker
>    #3742

端口环回已开启，docker卸载了，还是一样

coolsnowwolf, 2020-06-16:
不是卸载 docker，而是别编译进去

hxlls, 2020-06-16:
如果是X86 ，以上操作建议再加一个。关闭Flow Offloading，你就会发现问题都解决了

zeruns, 2020-06-16:
> 不是卸载 docker，而是别编译进去

重新编译了，没有把docker编译进去，结果一样

zeruns, 2020-06-16:
用别人旧版的固件倒没有这个问题

hxlls, 2020-06-16:
rm .config 后重新选择，不能选上docker 选上后再取消，依赖没有取消的，你怎么编译都没用

zeruns, 2020-06-16:
> rm .config 后重新选择，不能选上docker 选上后再取消，依赖没有取消的，你怎么编译都没用

我是直接把整个lede文件夹删掉然后重新git，重新编译

zeruns, 2020-06-17:
在gitee上找到别人fork的R20.4.8版本的编译后就没问题了，确实是新版的BUG

WXfeng-boy, 2020-06-18:
最新版没问题的，我一直没有编译docker。删除lede，重新git，重新配置config，全新刷机，别保留配置试试吧

ptsa, 2020-06-18:
> 如果是X86 ，以上操作建议再加一个。关闭Flow Offloading，你就会发现问题都解决了

Flow Offloading x86 以前好像没什么问题， 新版好像不行。 对x86 在没bug的情况下有作用么

zeruns, 2020-06-18:
> > 如果是X86 ，以上操作建议再加一个。关闭Flow Offloading，你就会发现问题都解决了
> 
> Flow Offloading x86 以前好像没什么问题， 新版好像不行。 对x86 在没bug的情况下有作用么

关闭结果一样

lePig, 2020-06-19:
同样遇到内网无法通过域名访问的问题。 大家上面说的都试了 好像还是不行

zeruns, 2020-06-19:
> 同样遇到内网无法通过域名访问的问题。 大家上面说的都试了 好像还是不行

我把R20.4.8版的编译了上传上来了，这个版本没问题：https://blog.zeruns.tech/archives/562.html

lePig, 2020-06-19:
@zeruns  好的谢谢。 我晚点看看。想问一下，你编译的这个固件除了酸酸乳，还有哪些插件？因为我只需要酸酸乳，如果插件太多 就不准备刷了。

zeruns, 2020-06-19:
> @zeruns 好的谢谢。 我晚点看看。想问一下，你编译的这个固件除了酸酸乳，还有哪些插件？因为我只需要酸酸乳，如果插件太多 就不准备刷了。

[image]

lePig, 2020-06-19:
@zeruns Thinks

zeruns, 2020-06-19:
> @zeruns Thanks

看评论区，文章中的是有问题的新版

ivybridgei, 2020-06-22:
不止是软路由，我用硬路由也遇到这个问题了，5月的源码正常使用，6月末的有问题

closed 2020-07-04 by coolsnowwolf