coolsnowwolf/lede #4821: Netflix 分流规则内Hulu无法使用

assist726, 2020-06-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用Netflix分流服务器无法打开Hulu视频

发现规则内Hulu的域名不全所以根据自己手机上surfboard的debug又加了一些域名还是不可以:

hulu.com
hulustream.com
huluim.com
hulu.hb.omtrdc.net
cws-hulu.conviva.com
imrworldwide.com
tealiumiq.com
tiqcdn.com
conviva.com
happyon.jp
hulu.jp

是否还需要增加更多分流IP规则?

coolsnowwolf, 2020-06-16:
没条件测试

assist726, 2020-06-17:
后来又测试了一下发现网页版可以正确访问了，但是手机APP估计有什么额外的地理位置验证吧

coolsnowwolf, 2020-06-17:
手机APP有可能是通过 httpdns 获取ip 的，这种情况就比较复杂了，需要把相应的 CDN 的 IP 段也加入。Netflix 的盒子 APP 就是这样的情况

assist726, 2020-06-17:
@coolsnowwolf 明白了谢谢，nslookup 去找IP即可？ 我倒是提交个PR吧

closed 2020-06-17 by assist726

Deckard-f-Cain, 2021-02-06:
> @coolsnowwolf 明白了谢谢，nslookup 去找IP即可？ 我倒是提交个PR吧

老哥，能公布一下 hulu的分流规则，网址 及 IP吗？多谢！

hang1233, 2021-04-09:
hulu需要添加的分流
hulu.com
huluad.com
huluim.com
hulustream.com
Showtime
sho.com
showtime.com