coolsnowwolf/lede #4824: 希望大雕可以考虑一下适配r2s

dukh1216, 2020-06-16:
友善nanoPi R2S，之前大雕在群里也有提到过，这款arm开发板很适合小水管的环境，功耗低性能够用，OpenWrt官方已经适配，但是众所周知大家更喜欢大雕的op和一些插件，所以现在r2s编译固件比较麻烦，对新手很不友好，希望大雕可以考虑一下适配r2s，这样编译起来会很轻松，小白也可以愉快地编译大雕的固件。

hxlls, 2020-06-16:
[image]
https://github.com/coolsnowwolf/lede/issues/3031#issuecomment-585644511 大雕已经公开回复过了，为什么不先进行ISSUES 搜索？

dukh1216, 2020-06-16:
> [image]
> [#3031 (comment)](https://github.com/coolsnowwolf/lede/issues/3031#issuecomment-585644511) 大雕已经公开回复过了，为什么不先进行ISSUES 搜索？

现在openwrt官方已经支持了，所以我才问

closed 2020-06-16 by dukh1216