coolsnowwolf/lede #4825: 最新源码DHCP分配列表 显示异常

tone88, 2020-06-16:
DHCP分配列表 设备信息重复显示，如图：

[image]

liuyanantj, 2020-06-17:
的确 我的也是这种情况  暂时不影响上网  不知道什么时候能修复

swxk521, 2020-06-17:
最新版有异常

zjh1982, 2020-06-17:
刚想提问，原来不是我一个人的问题，不过开机时间长负载很高

WXfeng-boy, 2020-06-17:
一样的问题

helianling, 2020-06-17:
threads=$(cat /proc/cpuinfo | grep 'processor' | wc -l)

	for i in $(seq 1 $threads); do
    procd_open_instance $cfg$i
    procd_set_param command $PROG -C $CONFIGFILE -k -x /var/run/dnsmasq/dnsmasq."${cfg}"."$i".pid
    procd_set_param file $CONFIGFILE
    [ -n "$user_dhcpscript" ] && procd_set_param env USER_DHCPSCRIPT="$user_dhcpscript"
    procd_set_param respawn

    procd_add_jail dnsmasq ubus log
    procd_add_jail_mount $CONFIGFILE $TRUSTANCHORSFILE $HOSTFILE $RFC6761FILE $DHCPBOGUSHOSTNAMEFILE /etc/passwd /etc/group /etc/TZ /dev/null /dev/urandom $dnsmasqconffile $dnsmasqconfdir $resolvfile $user_dhcpscript /etc/hosts /etc/ethers /sbin/hotplug-call $EXTRA_MOUNT $DHCPSCRIPT
    procd_add_jail_mount_rw /var/run/dnsmasq/ $leasefile

    procd_close_instance
	done

jlong0983, 2020-06-18:
### 同样遇到这个问题，不影响使用

[image]

xiaoguoyi, 2020-06-18:
> 刚想提问，原来不是我一个人的问题，不过开机时间长负载很高

[image]
我这个占用很大，不知道为啥

jlong0983, 2020-06-18:
> > 刚想提问，原来不是我一个人的问题，不过开机时间长负载很高
> 
> [image]
> 我这个占用很大，不知道为啥

dnsmasq开多，nlbwmon的统计工作量增加了？猜的，不懂编程，留下了没技术的眼泪

lloyd18, 2020-06-18:
同样

closed 2020-06-21 by coolsnowwolf