coolsnowwolf/lede #4826: baidupcs-web源码已删库

github-xihe, 2020-06-16:
https://github.com/Erope/baidupcs-web  项目已删库
[image]

zerofo, 2020-06-16:
嗯，， 刚去看了    404
`2020-06-16T06:28:57.3976498Z SHELL= flock /home/runner/work/lede/lede/tmp/.BaiduPCS-Go-v3.7.1-linux-amd64.zip.flock -c '  	/home/runner/work/lede/lede/scripts/download.pl "/home/runner/work/lede/lede/dl" "BaiduPCS-Go-v3.7.1-linux-amd64.zip" "skip" "" "https://github.com/Erope/baidupcs-web/releases/download/3.7.1/"    '
2020-06-16T06:28:57.4245250Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://github.com/Erope/baidupcs-web/releases/download/3.7.1/BaiduPCS-Go-v3.7.1-linux-amd64.zip
2020-06-16T06:28:57.4320012Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2020-06-16T06:28:57.4326255Z                                  Dload  Upload   Total   Spent    Left  Speed
2020-06-16T06:28:57.4326397Z 
2020-06-16T06:28:57.5012392Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2020-06-16T06:28:57.5013711Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2020-06-16T06:28:57.5025277Z curl: (22) The requested URL returned error: 404 Not Found
2020-06-16T06:28:57.5041525Z Download failed.
2020-06-16T06:28:57.5132913Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/BaiduPCS-Go-v3.7.1-linux-amd64.zip
2020-06-16T06:28:57.5200573Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2020-06-16T06:28:57.5201113Z                                  Dload  Upload   Total   Spent    Left  Speed
2020-06-16T06:28:57.5201369Z 
2020-06-16T06:28:57.9231041Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2020-06-16T06:28:57.9232397Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2020-06-16T06:28:57.9239671Z curl: (22) The requested URL returned error: 404 Not Found
2020-06-16T06:28:57.9260440Z Download failed.
2020-06-16T06:28:57.9353083Z + curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/BaiduPCS-Go-v3.7.1-linux-amd64.zip
2020-06-16T06:28:57.9436015Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2020-06-16T06:28:57.9441173Z                                  Dload  Upload   Total   Spent    Left  Speed
2020-06-16T06:28:57.9441449Z 
2020-06-16T06:28:58.2705913Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2020-06-16T06:28:58.4704633Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2020-06-16T06:28:58.4705606Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2020-06-16T06:28:58.4712007Z curl: (22) The requested URL returned error: 404 Not Found
2020-06-16T06:28:58.4725596Z Download failed.
2020-06-16T06:28:58.4727123Z No more mirrors to try - giving up.
2020-06-16T06:28:58.4736707Z make[3]: *** [/home/runner/work/lede/lede/dl/BaiduPCS-Go-v3.7.1-linux-amd64.zip] Error 2
2020-06-16T06:28:58.4737470Z Makefile:88: recipe for target '/home/runner/work/lede/lede/dl/BaiduPCS-Go-v3.7.1-linux-amd64.zip' failed
2020-06-16T06:28:58.4738528Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/lean/baidupcs-web'
2020-06-16T06:28:58.4747457Z time: package/lean/baidupcs-web/compile#0.18#0.07#1.22
2020-06-16T06:28:58.4748213Z package/Makefile:111: recipe for target 'package/lean/baidupcs-web/compile' failed
2020-06-16T06:28:58.4749473Z make[2]: *** [package/lean/baidupcs-web/compile] Error 2
2020-06-16T06:28:58.4773166Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2020-06-16T06:28:58.4787387Z package/Makefile:107: recipe for target '/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
2020-06-16T06:28:58.4788763Z make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2020-06-16T06:28:58.4808635Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2020-06-16T06:28:58.4824456Z make: *** [world] Error 2
2020-06-16T06:28:58.4824990Z /home/runner/work/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed
2020-06-16T06:28:58.4915581Z ##[error]Process completed with exit code 2.`

eallion, 2020-06-17:
不用百度盘就不会有这种烦恼。

goofysun, 2020-06-17:
不用不行啊，小水管只有路由器程序才适合上传下载，开着电脑台折磨人了。5T大空间，省了一个硬盘了

aizhouyi, 2020-06-17:
太难了，回退另外的baidupcs-web版本登不上，一直转圈圈。

chanwang5, 2020-06-18:
github上搜索 找到另一个 然后 自己改或者替换文件看看
https://github.com/garypang13

liwenjie119, 2020-06-19:
之前没fork，现在已经没了

guochan2019, 2020-06-19:
以前有dl的3.7.1版，还是可以编译的

aepub, 2020-06-20:
难怪make download时就在这个地方出现错误，不能继续下去，luci里面没有选择baidupcs-web也不行！

Michaol, 2020-06-23:
难怪今天本地编译，云编译就错误。

aizhouyi, 2020-06-24:
博客地址还有v3.7.1的二进制文件，已经有人提交pr了

yearqp, 2020-06-24:
https://github.com/liuzhuoling2011/baidupcs-web
这个还没有删啊

tonc, 2020-06-26:
https://www.shinenet.cn/down/

Erope, 2020-06-26:
嗯 前端仓库(托管release)已经删库，后端仓库(可直接编译)https://github.com/Erope/BaiduPCS-Go 还没删，不过也停止维护了。
删库是跟随原作者iikira删库而删库的，不想为原作者带来更多的困扰，也是为我自己考虑。
目前博客的下载地址仍然可用，近日会做一个301重定向到由CloudFlare托管的地址，以减少博客GCP的流量支出。

Erope, 2020-06-26:
不过也仍然推荐尽快更换主用网盘吧。
不用百度盘就不会有这种烦恼。

liwenjie119, 2020-06-26:
> 不过也仍然推荐尽快更换主用网盘吧。
> 不用百度盘就不会有这种烦恼。

刚开一年会员不到两个月，用这个其实是为了挂软路由下载，不用开电脑挂机了。这个凉凉以后，不知道还有什么能挂机的方案

closed 2020-06-29 by github-xihe