coolsnowwolf/lede #4831: 是否应该现在跟进升级到 5.4 内核的意见收集

coolsnowwolf, 2020-06-17:
关于是否应该现在跟进升级到 5.4 内核的意见收集

升级到 5.4 内核将获得：
1. 更及时的内核补丁支持
2. 新硬件的驱动，例如 AMD Zen2 APU 的 GPU 解码，2.5Gbps 各种网卡的支持等
3. 新的 DSA 网卡模型，例如 MT7503 可以强制指定每一个端口的速率解决掉线问题
.... 等等

升级到 5.4 内核将失去的：
1. 效能。 5.4 内核的 Flow 比 4.19 慢了 10-18% 以上，低端硬件特别明显。
2. 闭源驱动，例如 K2P 的驱动并没有在 5.4 内核中被支持
3. 更多的 Bug ，因为 5.4 不是正式 LTS 内核，也没有 stable release 的 OpenWrt 源码

PS: 5.4 内核寿命特别短，明年12月就终结了

大家意见如何？

EuropeanWithSSR, 2020-06-17:
不要，k2p再用5年

frely, 2020-06-17:
没有新硬件，不更新。

5t-RawBeRry, 2020-06-17:
做个 5.4  的分支？

kisaragychihaya, 2020-06-17:
反对，稳定好用才是最要紧

benchzjb, 2020-06-17:
不追新，路由硬件更换速度不高

tianmotrue, 2020-06-17:
不要。反对！不更

xqiushi, 2020-06-17:
不要，稳定才好。

305267382, 2020-06-17:
等LTS的内核出来再升级

naoki66, 2020-06-17:
x64 x86 支持

AmadeusGhost, 2020-06-17:
个人不建议现在就升级，kernel 5.4 需要 mac80211 5.4 及以上，然而官方现在的无线稳定性和最新 5.7-rc3 的性能都不大好
不说无线的话，其实 5.4 内核在 ipq40xx/x86 上用着也没啥差

tjlive, 2020-06-17:
反对，或者做个 5.4 的分支

scllen, 2020-06-17:
不建议更新 5.4 内核

martin0607, 2020-06-17:
单独x86更新5.4？

liuyanantj, 2020-06-17:
不建议跟新  没有这方面的需求  大家的网络是不是也没有达到这个级别了

alvazhang79, 2020-06-17:
个人认为，稳定高效才是王道。

bobstone007, 2020-06-17:
当前版本很好，稳定又高效

Lasx, 2020-06-17:
x86的可以试试，硬路由就算了吧

hcym, 2020-06-17:
armv8的强烈推荐，x86无所谓，看大雕心情😙😂

sxf5487, 2020-06-17:
不建议更新，如果确实需要的话建议拉个分支。

zcluo, 2020-06-17:
x64 x86 支持

spacerip-per, 2020-06-17:
等下一个longterm版本再升级

ggllioio, 2020-06-17:
等下个lts内核

qyh214, 2020-06-17:
建议稳定后再更吧，可以放到一个新分支里，或单独一个库里，如果有尝鲜的可以在那里。

路由器在稳定的同时，再提供很棒的功能挺重要的，如果各种蛋疼事情，反而不好。

内核除非变成LTS支持，不然并不建议，不是大家的硬件都能这么快与时俱进的。

hxlls, 2020-06-17:
可以开test 分支。这样比较好，至于FLOW性能损失这个，因为Flow Offloading引起不少问题，所以我都是直接关闭的，也没看到对实际占用率有什么大影响，所以我认为暂时不需要考虑这个性能

aizhouyi, 2020-06-17:
不建议升级也不建议新分支，权衡利弊。真有拿最新硬件去玩openwrt的吗，估计非常少非常少。稳定压倒一切，升级非lts内核，升级出更多的bug。做精一个分支我就觉得很赞。

IDSSC, 2020-06-17:
与其维护一个不确定的内核，还不如花精力搞好现有的

duanego, 2020-06-17:
nonono

star-o0O, 2020-06-17:
应该上，不折腾和咸鱼有什么区别

Promix953, 2020-06-17:
支持X86上5.4
PS：感觉issues里征求这个意见基本就决定了否定的居多

mjyhj, 2020-06-17:
支持升级，关于那个效能平时都是提升50% 80% 100%什么的
现在才降低10-18%，丝毫不影响，下次再提升个50%即可

qxzg, 2020-06-17:
不支持，作为一个路由器系统，我认为稳定才是第一要务，其他绝大多数问题都需要给稳定性让路。
况且支持时间这么短，上的话还会花很多时间折腾，感觉有点得不偿失？

lxhao61, 2020-06-17:
反对，稳定好用才是最要紧!

lxhao61, 2020-06-17:
> 可以开test 分支。这样比较好，至于FLOW性能损失这个，因为Flow Offloading引起不少问题，所以我都是直接关闭的，也没看到对实际占用率有什么大影响，所以我认为暂时不需要考虑这个性能

就是！我从不编译Flow Offloading！

BOBINIUNIU, 2020-06-18:
虽然我用x86，性能啥的无所谓，不过建议还是等等再说，万一有内核bug很麻烦，牵一发而动全身。目前的4.19版本已经比较稳定了，想要折腾的自己开分支吧。

ihuaer, 2020-06-18:
路由器稳定最重要
新功能可以放近测试项目里

ITdesk01, 2020-06-18:
建议增加新分支，虽然内核时间较短，性能下降，但总体来说可以试试水，为以后新设备铺路，毕竟现在wifi6价格血拼如此低，可以预想的，未来5G 千兆 wifi6

KevinMX, 2020-06-18:
对于路由器来说，稳定好用才是最重要的，为了追新失去太多不太值得。对于部分设备有必要升级5.4内核的可以考虑单开新分支/单独支持（虽然这样会稍微增大工作量吧，各位辛苦了:）。
（另外，Linux 5.4 LTS不是到2024才会EOL吗？ Link: [The Linux Kernel Archives](https://www.kernel.org/category/releases.html)）

PandaEgg, 2020-06-18:
反对。稳定好用才是王道

jzhou404, 2020-06-18:
对于路由器来说，稳定好用才是最重要的. 不建议更新 5.4 内核

nivalxer, 2020-06-18:
建议等内核LTS。
不过可以考虑先建立分支，包括mac80211、DSA改动相对较大，升级需要花费时间进行适配，可先行在分支上进行测试。

maxjove, 2020-06-18:
不要

waynon, 2020-06-18:
https://www.kernel.org/category/releases.html
Linux内核官网上写的不是支持到2025年12月吗？
建议升级到5.4

shenzt68, 2020-06-18:
可以弄个5.4分支

ptsa, 2020-06-18:
> 可以弄个5.4分支

+1

ricksuzade-maker, 2020-06-18:
5.4是官方的LTS内核，稳定性应该没问题，放心用吧

pirately, 2020-06-19:
支持升级

ricksuzade-maker, 2020-06-19:
其实理由只有一条，koolshare已经升级5.4了，再不升就被超过了。

xtwz, 2020-06-19:
不要，稳定才好。

stock169, 2020-06-19:
升吧

Jefffish09, 2020-06-19:
不更新。稳定是第一位。要折腾or尝新，可以新开分支。主分支还是要稳住。

ricksuzade-maker, 2020-06-19:
都玩软路由了，何必在折腾硬路由，直接用原厂固件不就行了吗

liwenjie119, 2020-06-19:
5.4的生命周期延长到2025年，但是仍然不是lts

mumuou, 2020-06-19:
X86可以单独搞一个选项到5.4，其它的没有必要~~~

biozzl, 2020-06-20:
没有新硬件，不更新。

lqh280, 2020-06-20:
维稳重要，不追新~辛苦了

Biglilo, 2020-06-20:
5.4是长期更新版本，升噻。

感谢L大和各位开发人员的努力。

OldTaoge, 2020-06-21:
5.4虽然有很多新特性，但Openwrt项目是针对嵌入式设备的Linux操作系统。
但现在嵌入式设备的性能以及一代机皇K2P的原因，加之支持很短，付出的精力和收益不成正比

zcluo, 2020-06-21:
> 5.4是长期更新版本，升噻。
> 
> 感谢L大和各位开发人员的努力。

好像续命了

xakaff, 2020-06-21:
本人亲测 lienaol 5.4.32 master内核三个月使用体验，非上面无脑评论：
1、5.4内核使用非常稳定，无内存泄漏及长期使用网络延迟等不稳定现象，master分支已经非常成熟，为何不更新；
2.、5.4内核比4.19内核的cpu占用率和温度下降20%，4.19内核无主动散热温度50度时，5.4内核在42度左右，新内核效能机制已经体现无疑；
3、linux官方5.4内核已经是长期支持版本，连ubuntu和PVE都已升为此版本或更高版本，为何众人还是痴迷于老版本不做改变，并且我的OMV主机已经官方升为5.6内核并使用非常稳定，openwrt的使用者都是以最新为乐，稳定的同时最求最佳性能也是主要的目标，L大也是新技术的倡导者，个人建议还是有最新内核技术为佳。谢谢L大长期以来给我们带来最新openwrt的体验。谢谢

xakaff, 2020-06-21:
补充：本人四台路由主机，E2 1275L v3 、E3 1230L v3、J3160、J1900 这四台用在我的四套房子里，不做pve虚拟，单独主机使用。之所以四台主机一直使用lienol固件就是因为L大的固件是4,19，原因就如上面所述。如果大家对lienol master分支固件最后一版有优化就知道其固件的优势，虽然目前早已停止更新。

[80 more replies]

closed 2020-07-09 by coolsnowwolf