coolsnowwolf/lede #4838: 希望保留pptp vpn server及修改ddns

lqh280, 2020-06-19:
想自己编译固件，发现：
1、没见到pptp vpn server，是精简了？希望能保留，这个对我这种菜鸟方便，简单设置就能连进家里的网络。
2、DDNS，我用的花生壳，好像默认定时10分钟，也不会自动更新新的IP。我用WinSCP，修改了文件： /etc/init.d/ddns

#!/bin/sh /etc/rc.common
START=99  #原来是95，网上有教程说修改99，我必了后，应该是正常了。

zeruns, 2020-06-19:
可以用IPSec VPN

lqh280, 2020-06-19:
IPSec VPN我设置了，二个一样的密码，Windows添加VPN后，更改了VPN类型为“使用IPSec的第2层隧道协议”，但不知道是哪没设置了，连接不上。

lqh280, 2020-06-19:
> 可以用IPSec VPN

IPSec VPN我设置了，二个一样的密码，Windows添加VPN后，更改了VPN类型为“使用IPSec的第2层隧道协议”，但不知道是哪没设置了，连接不上。

zeruns, 2020-06-19:
> > 可以用IPSec VPN
> 
> IPSec VPN我设置了，二个一样的密码，Windows添加VPN后，更改了VPN类型为“使用IPSec的第2层隧道协议”，但不知道是哪没设置了，连接不上。

windows不支持这个，安卓和IOS都可以

lqh280, 2020-06-19:
> > > 可以用IPSec VPN
> > 
> > 
> > IPSec VPN我设置了，二个一样的密码，Windows添加VPN后，更改了VPN类型为“使用IPSec的第2层隧道协议”，但不知道是哪没设置了，连接不上。
> 
> windows不支持这个，安卓和IOS都可以

明白了，谢谢

closed 2020-07-04 by coolsnowwolf