coolsnowwolf/lede #4845: SSR Plus+ gfw模式变成全局代理

yantoumu, 2020-06-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
SSR Plus+ 

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
SSR Plus+ 模式设置为gfw，但是会变成全局代理，访问国内的都会跑到ssr的服务器上，v2ray,trojan都是这样的情况，

### (1) 具体问题

A：

### (2) 路由器型号和固件版本 

A：  固件版本 (git-20.088.29068-cebe120) / OpenWrt R20.4.8 
        主机型号 | Intel(R) Celeron(R) CPU J1900 @ 1.99GHz : 4 Core 4 Thread

### (3) 详细日志

A：
Sat Jun 20 12:08:41 2020 kern.notice kernel: [    0.038851] random: get_random_bytes called from start_kernel+0x85/0x543 with crng_init=0
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.038866] setup_percpu: NR_CPUS:8 nr_cpumask_bits:8 nr_cpu_ids:4 nr_node_ids:1
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.039584] percpu: Embedded 41 pages/cpu s128280 r8192 d31464 u524288
Sat Jun 20 12:08:41 2020 kern.debug kernel: [    0.039601] pcpu-alloc: s128280 r8192 d31464 u524288 alloc=1*2097152
Sat Jun 20 12:08:41 2020 kern.debug kernel: [    0.039604] pcpu-alloc: [0] 0 1 2 3
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.039645] Built 1 zonelists, mobility grouping on.  Total pages: 488400
Sat Jun 20 12:08:41 2020 kern.notice kernel: [    0.039650] Kernel command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=234050d5-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.040296] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.040587] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
Sat Jun 20 12:08:41 2020 kern.debug kernel: [    0.040654] Calgary: detecting Calgary via BIOS EBDA area
Sat Jun 20 12:08:41 2020 kern.debug kernel: [    0.040657] Calgary: Unable to locate Rio Grande table in EBDA - bailing!
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.053844] Memory: 1929812K/1984916K available (10248K kernel code, 560K rwdata, 2044K rodata, 1068K init, 2388K bss, 55104K reserved, 0K cma-reserved)
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.053941] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.054022] rcu: Hierarchical RCU implementation.
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.054026] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.054030] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.054034] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.054053] NR_IRQS: 4352, nr_irqs: 1024, preallocated irqs: 16
Sat Jun 20 12:08:41 2020 kern.debug kernel: [    0.054407] spurious 8259A interrupt: IRQ7.
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.056094] Console: colour VGA+ 80x25
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.063948] console [tty0] enabled
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.095684] console [ttyS0] enabled
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.095943] ACPI: Core revision 20180810
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.096395] APIC: Switch to symmetric I/O mode setup
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.097579] ..TIMER: vector=0x30 apic1=0 pin1=2 apic2=0 pin2=0
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.146419] clocksource: tsc-early: mask: 0xffffffffffffffff max_cycles: 0x39a275294b2, max_idle_ns: 881590412759 ns
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.147044] Calibrating delay loop (skipped), value calculated using timer frequency.. 3998.40 BogoMIPS (lpj=19992000)
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.147675] pid_max: default: 32768 minimum: 301
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.148038] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.148456] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
Sat Jun 20 12:08:41 2020 kern.warn kernel: [    0.149160] ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
Sat Jun 20 12:08:41 2020 kern.warn kernel: [    0.149536] ENERGY_PERF_BIAS: View and update with x86_energy_perf_policy(8)
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.149984] CPU0: Thermal monitoring enabled (TM1)
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.150301] process: using mwait in idle threads
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.150609] Last level iTLB entries: 4KB 48, 2MB 0, 4MB 0
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.150953] Last level dTLB entries: 4KB 128, 2MB 16, 4MB 16, 1GB 0
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.151340] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.151864] Spectre V2 : Mitigation: Full generic retpoline
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.152216] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.152724] Spectre V2 : Enabling Restricted Speculation for firmware calls
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.153145] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.153663] MDS: Vulnerable: Clear CPU buffers attempted, no microcode
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.154251] Freeing SMP alternatives memory: 28K
Sat Jun 20 12:08:41 2020 kern.debug kernel: [    0.157564] TSC deadline timer enabled
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.157568] smpboot: CPU0: Intel(R) Celeron(R) CPU  J1900  @ 1.99GHz (family: 0x6, model: 0x37, stepping: 0x9)
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.158471] Performance Events: PEBS fmt2+, 8-deep LBR, Silvermont events, 8-deep LBR, full-width counters, Intel PMU driver.
Sat Jun 20 12:08:41 2020 kern.info kernel: [    0.159227] ... version:                3
[... 589 lines cut ...]
Sat Jun 20 12:13:39 2020 daemon.warn pdnsd[8776]: Caught signal 15. Exiting.
Sat Jun 20 12:13:39 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 20 12:13:39 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 20 12:13:42 2020 cron.info crond[9624]: crond (busybox 1.30.0) started, log level 8
Sat Jun 20 12:13:43 2020 daemon.info pdnsd[9842]: pdnsd-1.2.9b-par starting.
Sat Jun 20 12:13:44 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 20 12:13:44 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 20 12:14:14 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 20 12:14:14 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 20 12:14:14 2020 daemon.err uhttpd[2681]: udhcpc: started, v1.30.0
Sat Jun 20 12:14:14 2020 daemon.err uhttpd[2681]: udhcpc: sending discover
Sat Jun 20 12:14:17 2020 daemon.err uhttpd[2681]: udhcpc: no lease, failing
Sat Jun 20 12:14:17 2020 daemon.warn pdnsd[9842]: Caught signal 15. Exiting.
Sat Jun 20 12:14:18 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 20 12:14:18 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 20 12:14:21 2020 cron.info crond[10480]: crond (busybox 1.30.0) started, log level 8
Sat Jun 20 12:14:23 2020 daemon.info pdnsd[10697]: pdnsd-1.2.9b-par starting.
Sat Jun 20 12:14:24 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 20 12:14:24 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 20 12:14:37 2020 daemon.warn pdnsd[10697]: Caught signal 15. Exiting.
Sat Jun 20 12:14:37 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jun 20 12:14:40 2020 cron.info crond[11079]: crond (busybox 1.30.0) started, log level 8
Sat Jun 20 12:14:41 2020 daemon.info pdnsd[11297]: pdnsd-1.2.9b-par starting.
Sat Jun 20 12:14:42 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jun 20 12:14:42 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

Henshufu, 2020-06-20:
我也遇到过,passwall也有这个情况。

yantoumu, 2020-06-20:
> 我也遇到过，passwall也有这个情况。

也就是最近才出现的，之前好好的

cnekol, 2020-06-22:
同上，看到这个issue就去看了一眼，确实有不少网站是跑的代理。不过我是用的大陆白名单。

yantoumu, 2020-06-25:
还有人遇到同样情况的吗？？

closed 2020-07-01 by coolsnowwolf