coolsnowwolf/lede #4847: 新编译的固件重启后，拨号成功不能上网，DNS解析不管用，不知道是BUG还是我编译有问题

nj123456, 2020-06-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：failed to create inotify for /tmp/resolv.conf.auto: No space left on device
日志里面显示这个
failed to create inotify for /tmp/resolv.conf.auto: No space left on device

nj123456, 2020-06-20:
Sun Jun 21 00:17:44 2020 daemon.err block: mounting /dev/sda3 (ext4) as /mnt/sda3 failed (16) - Not a tty
Sun Jun 21 00:17:48 2020 daemon.err collectd[3449]: Filter subsystem: Built-in target `write': Dispatching value to all write plugins failed with status -1.
Sun Jun 21 00:17:48 2020 user.notice mwan3[7765]: Execute ifup event on interface wan (pppoe-wan)
Sun Jun 21 00:17:48 2020 user.notice mwan3[7765]: Starting tracker on interface wan (pppoe-wan)
Sun Jun 21 00:17:48 2020 user.info mwan3[7765]: connection tracking not flushed on interface wan (pppoe-wan) ifup
Sun Jun 21 00:17:48 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Jun 21 00:17:49 2020 user.notice ddns-scripts[8431]: myddns_ipv4: PID '8431' started at 2020-06-21 00:17
Sun Jun 21 00:17:59 2020 daemon.crit dnsmasq[12867]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:17:59 2020 daemon.crit dnsmasq[12867]: FAILED to start up
Sun Jun 21 00:17:59 2020 daemon.crit dnsmasq[12866]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:17:59 2020 daemon.crit dnsmasq[12866]: FAILED to start up
Sun Jun 21 00:18:04 2020 daemon.crit dnsmasq[12939]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:04 2020 daemon.crit dnsmasq[12939]: FAILED to start up
Sun Jun 21 00:18:04 2020 daemon.crit dnsmasq[12938]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:04 2020 daemon.crit dnsmasq[12938]: FAILED to start up
Sun Jun 21 00:18:04 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sun Jun 21 00:18:07 2020 user.info mwan3track[8106]: Check (ping) failed for target "www.baidu.com" on interface wan (pppoe-wan)
Sun Jun 21 00:18:09 2020 daemon.crit dnsmasq[13235]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:09 2020 daemon.crit dnsmasq[13235]: FAILED to start up
Sun Jun 21 00:18:09 2020 daemon.crit dnsmasq[13236]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:09 2020 daemon.crit dnsmasq[13236]: FAILED to start up
Sun Jun 21 00:18:14 2020 daemon.crit dnsmasq[13259]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:14 2020 daemon.crit dnsmasq[13259]: FAILED to start up
Sun Jun 21 00:18:14 2020 daemon.crit dnsmasq[13258]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:14 2020 daemon.crit dnsmasq[13258]: FAILED to start up
Sun Jun 21 00:18:18 2020 user.info mwan3track[8106]: Check (ping) failed for target "www.baidu.com" on interface wan (pppoe-wan)
Sun Jun 21 00:18:19 2020 daemon.crit dnsmasq[13301]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:19 2020 daemon.crit dnsmasq[13301]: FAILED to start up
Sun Jun 21 00:18:19 2020 daemon.crit dnsmasq[13302]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:19 2020 daemon.crit dnsmasq[13302]: FAILED to start up
Sun Jun 21 00:18:20 2020 daemon.info pppd[2936]: System time change detected.
Sun Jun 21 00:18:23 2020 daemon.err collectd[3449]: Filter subsystem: Built-in target `write': Some write plugin is back to normal operation. `write' succeeded.
Sun Jun 21 00:18:24 2020 daemon.crit dnsmasq[13327]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:24 2020 daemon.crit dnsmasq[13327]: FAILED to start up
Sun Jun 21 00:18:24 2020 daemon.crit dnsmasq[13326]: failed to create inotify for /tmp/resolv.conf.auto: No space left on device
Sun Jun 21 00:18:24 2020 daemon.crit dnsmasq[13326]: FAILED to start up
Sun Jun 21 00:18:24 2020 daemon.info procd: Instance dnsmasq::cfg01411c1 s in a crash loop 6 crashes, 0 seconds since last crash
Sun Jun 21 00:18:24 2020 daemon.info procd: Instance dnsmasq::cfg01411c2 s in a crash loop 6 crashes, 0 seconds since last crash
Sun Jun 21 00:18:30 2020 user.info mwan3track[8106]: Check (ping) failed for target "www.baidu.com" on interface wan (pppoe-wan)

goldkeyber112, 2020-06-20:
failed to create inotify for /tmp/resolv.conf.auto: No space left on device

closed 2020-07-03 by coolsnowwolf