coolsnowwolf/lede #4851: 编译orangepi pc的时候出错

njlees, 2020-06-21:
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target '/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/r8168-8.048.03/r8168_n.o' failed
make[6]: *** [/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/r8168-8.048.03/r8168_n.o] Error 1
Makefile:1520: recipe for target '_module_/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/r8168-8.048.03' failed
make[5]: *** [_module_/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/r8168-8.048.03] Error 2
make[5]: Leaving directory '/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/linux-4.19.122'
Makefile:140: recipe for target 'modules' failed
make[4]: *** [modules] Error 2
make[4]: Leaving directory '/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/r8168-8.048.03'
Makefile:55: recipe for target '/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/r8168-8.048.03/.built' failed
make[3]: *** [/home/king/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/r8168-8.048.03/.built] Error 2
make[3]: Leaving directory '/home/king/lede/package/lean/r8168'
time: package/lean/r8168/compile#1.98#0.26#2.44
package/Makefile:111: recipe for target 'package/lean/r8168/compile' failed
make[2]: *** [package/lean/r8168/compile] Error 2
make[2]: Leaving directory '/home/king/lede'
package/Makefile:107: recipe for target '/home/king/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/king/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/king/lede'
/home/king/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

每次都是到这里出错

AmadeusGhost, 2020-06-22:
把`kmod-r8168`去掉，之前那个 pr 直接所有设备默认选上了 emmm

jackiezhang29, 2020-06-22:
> 把`kmod-r8168`去掉，之前那个 pr 直接所有设备默认选上了 emmm

x86也出现了同样的问题

closed 2020-07-04 by coolsnowwolf