coolsnowwolf/lede #4854: luci-app-v2ray-server能否添加tls支持

Plume-yu, 2020-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Vps跑的服务端 server版本只支持ws不支持tls 需求tls，

### (1) 具体问题
没tls 秒封 尴尬
A：

### (2) 路由器型号和固件版本
阿里云ecs

OpenWrt R20.6.18 / LuCI Master (git-20.141.34051-f750b5f)

A：

### (3) 详细日志

A：

9l, 2020-06-24:
tls 请装 nginx 或  caddy

yhl10000, 2020-06-24:
题主问的应该是client端吧。服务器端安装nginx什么的。
我去年编译额openwrt里面的V2RA Y确实不支持tls。

Plume-yu, 2020-06-24:
> 服务器端安装nginx什么的。
> 我去年编译额openwrt里面的V2RA Y确实不支持tls。

不是的 是服务端 我18号编译的固件上面的v2ray server 只有ws支持 并没有tls

9l, 2020-06-25:
> 题主问的应该是client端吧。服务器端安装nginx什么的。
> 我去年编译额openwrt里面的V2RA Y确实不支持tls。

呵呵

9l, 2020-06-25:
> > 服务器端安装nginx什么的。
> > 我去年编译额openwrt里面的V2RA Y确实不支持tls。
> 
> 不是的 是服务端 我18号编译的固件上面的v2ray server 只有ws支持 并没有tls

服务器 v2ray ws + nginx/caddy tls 需要域名

closed 2020-07-02 by coolsnowwolf