coolsnowwolf/lede #4855: 请Lede大神，早日支持RTL8125B与RTL8125芯片驱动(2.5G网卡，好多主板都集成）

wzbbstx, 2020-06-22:
请Lede大神，早日支持RTL8125B与RTL8125芯片驱动(2.5G网卡，好多主板都集成）

驱动地址如下：
https://www.realtek.com/zh-tw/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software

martin0607, 2020-06-22:
就算不集成自己也可以下载替换啊

zeruns, 2020-06-22:
> 就算不集成自己也可以下载替换啊

可以参考我这篇文章编译其他驱动进去：https://blog.zeruns.tech/archives/565.html

zeruns, 2020-06-22:
可以参考我这篇文章编译其他驱动进去：https://blog.zeruns.tech/archives/565.html

zeruns, 2020-06-22:
> 请Lede大神，早日支持RTL8125B与RTL8125芯片驱动(2.5G网卡，好多主板都集成）
> 
> 驱动地址如下：
> https://www.realtek.com/zh-tw/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software

可以参考我这篇文章编译其他驱动进去：https://blog.zeruns.tech/archives/565.html
文章中的git驱动的地址可以在GitHub搜索这款驱动的源码地址替换上去，刚刚我搜过有的

wzbbstx, 2020-06-22:
果然是大神。不过俺是编程苦手啊。。。能不能辛苦把8125B的驱动加入呀（之前的8125老驱动无法正常驱动8125B芯片），谢谢了。

zeruns, 2020-06-22:
> 果然是大神。不过俺是编程苦手啊。。。能不能辛苦把8125B的驱动加入呀（之前的8125老驱动无法正常驱动8125B芯片），谢谢了。
> […](#)

详细教程都给出来了，自己动手

wzbbstx, 2020-06-22:
好吧。大神。这两天我折腾下。。。

  

  
详细教程都给出来了，自己动手

helianling, 2020-06-23:
现在代码中的驱动是去年的，只支持8125（A），不支持最新的8125B，把图片中的文件用官网最新的Linux驱动源码（同时支持8125和8125B）全部替换就好了，外层的Makefile建议也改成最新驱动版本号。
[image]

wzbbstx, 2020-06-23:
> 现在代码中的驱动是去年的，只支持8125（A），不支持最新的8125B，把图片中的文件用官网最新的Linux驱动源码（同时支持8125和8125B）全部替换就好了，外层的Makefile建议也改成最新驱动版本号。
> [image]

谢谢高手指点。晚上回家折腾！！！

crosys, 2020-06-23:
> > 请Lede大神，早日支持RTL8125B与RTL8125芯片驱动(2.5G网卡，好多主板都集成）
> > 驱动地址如下：
> > https://www.realtek.com/zh-tw/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software
> 
> 可以参考我这篇文章编译其他驱动进去：https://blog.zeruns.tech/archives/565.html
> 文章中的git驱动的地址可以在GitHub搜索这款驱动的源码地址替换上去，刚刚我搜过有的

请教大大，RTL8156的2.5G USB网卡驱动按教程能搞定吗？

goldkeyber112, 2020-06-25:
已增加支持
#4866

closed 2020-07-02 by coolsnowwolf

laaArmy, 2021-09-15:
你好好看看人家叫啥