coolsnowwolf/lede #4857: 树莓派4B DDNS显示错误

yocosweet, 2020-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：树莓派4B DDNS显示错误

### (2) 路由器型号和固件版本

A：树莓派4B OpenWrt R20.6.18

### (3) 详细日志

A：
[image]

RainCat1998, 2020-06-22:
软路由同样遇到这个问题

RainCat1998, 2020-06-22:
`500 内部服务器错误
对不起，服务器遇到未知错误。

/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/ddns'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/tblsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'ddns/overview_doubleline'.
A runtime error occured: /usr/lib/lua/luci/model/cbi/ddns/overview.lua:99: bad argument #2 to 'tonumber' (integer expected, got string)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function 
Powered by LuCI Master (git-20.141.34051-f750b5f) / OpenWrt R20.6.18 / Tracemouse(theme)`

a1093773736, 2020-06-22:
同软路由器也是这问题

a1093773736, 2020-06-22:
但之前设置的DNS还是生效的。

RainCat1998, 2020-06-22:
> 但之前设置的DNS还是生效的。

看最新的issue，貌似网络的luci界面也出问题了，也是类似的错误

tonyliangli, 2020-06-22:
相同问题

yocosweet, 2020-06-23:
最新的源码已解决

closed 2020-06-23 by yocosweet