coolsnowwolf/lede #4858: 停用WiFi导致luci显示异常

Hey-Leo, 2020-06-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

 HiWiFi HC5962
OpenWrt R20.6.18，今晚（6月23）新编译的固件

停用2.4G无线后状态一直停留在 正在收集数据
[image]
[image]

### (3) 详细日志
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute firstchild dispatcher target for entry '/admin'.
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/status'.
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute template dispatcher target for entry '/admin/status/overview'.
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_status/index'.
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: A runtime error occured: /usr/lib/lua/luci/model/network.lua:1601: bad argument #2 to 'tonumber' (integer expected, got string)
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: stack traceback:
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: 	[C]: in function 'assert'
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
Tue Jun 23 00:13:22 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/'.
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute firstchild dispatcher target for entry '/admin'.
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/status'.
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute template dispatcher target for entry '/admin/status/overview'.
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_status/index'.
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: A runtime error occured: /usr/lib/lua/luci/model/network.lua:1601: bad argument #2 to 'tonumber' (integer expected, got string)
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: stack traceback:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: 	[C]: in function 'assert'
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute call dispatcher target for entry '/admin/network/wireless_status/radio0.network1,radio1.network1'.
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/model/network.lua:1601: bad argument #2 to 'tonumber' (integer expected, got string)
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: stack traceback:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: 	[C]: in function 'assert'
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
Tue Jun 23 00:13:33 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute call dispatcher target for entry '/admin/network/wireless_status/radio0.network1,radio1.network1'.
Tue Jun 23 00:13:33 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:13:33 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/model/network.lua:1601: bad argument #2 to 'tonumber' (integer expected, got string)
Tue Jun 23 00:13:33 2020 daemon.err uhttpd[1984]: stack traceback:
Tue Jun 23 00:13:33 2020 daemon.err uhttpd[1984]: 	[C]: in function 'assert'

前面的没了，现在日志就是这些一直重复

Hey-Leo, 2020-06-22:
补一下，重置后再关2.4GWiFi是出现的日志
Tue Jun 23 00:26:12 2020 kern.info kernel: [  268.319089] device wlan0 left promiscuous mode
Tue Jun 23 00:26:12 2020 kern.info kernel: [  268.323835] br-lan: port 3(wlan0) entered disabled state
Tue Jun 23 00:26:12 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Tue Jun 23 00:26:12 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Tue Jun 23 00:26:12 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue Jun 23 00:26:12 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Jun 23 00:26:12 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Tue Jun 23 00:26:12 2020 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Tue Jun 23 00:26:13 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute call dispatcher target for entry '/admin/network/wireless_status/radio0.network1,radio1.network1'.
Tue Jun 23 00:26:13 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:26:13 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/model/network.lua:1601: bad argument #2 to 'tonumber' (integer expected, got string)
Tue Jun 23 00:26:13 2020 daemon.err uhttpd[1984]: stack traceback:
Tue Jun 23 00:26:13 2020 daemon.err uhttpd[1984]: 	[C]: in function 'assert'
Tue Jun 23 00:26:13 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
Tue Jun 23 00:26:13 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
Tue Jun 23 00:26:18 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/dispatcher.lua:381: Failed to execute call dispatcher target for entry '/admin/network/wireless_status/radio0.network1,radio1.network1'.
Tue Jun 23 00:26:18 2020 daemon.err uhttpd[1984]: The called action terminated with an exception:
Tue Jun 23 00:26:18 2020 daemon.err uhttpd[1984]: /usr/lib/lua/luci/model/network.lua:1601: bad argument #2 to 'tonumber' (integer expected, got string)
Tue Jun 23 00:26:18 2020 daemon.err uhttpd[1984]: stack traceback:
Tue Jun 23 00:26:18 2020 daemon.err uhttpd[1984]: 	[C]: in function 'assert'
Tue Jun 23 00:26:18 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
Tue Jun 23 00:26:18 2020 daemon.err uhttpd[1984]: 	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

RainCat1998, 2020-06-22:
其他插件也有类似的问题，比如DDNS

gsblay, 2020-06-22:
R7800同样问题，回退0326b4重新编译正常。

helianling, 2020-06-23:
DDNS也是相同的问题：
bad argument #2 to 'tonumber' (integer expected, got string)
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: stack traceback:
Tue Jun 23 00:13:27 2020 daemon.err uhttpd[1984]: 	[C]: in function 'assert'

closed 2020-06-23 by coolsnowwolf