coolsnowwolf/lede #4863: 编译报错

hplee0120, 2020-06-23:
请问下这是哪里出问题，还是按照之前的编译步骤，之前都是一次过的，这两天重装系统之后一直过不去，试过挺多办法了都没解决。

g++ -x c++  -g -O2   -I. -I. -I./config -DLOCALEDIR="\"/home/ubuntu/lede/staging                                                                                                                                                             _dir/toolchain-x86_64_gcc-7.5.0_musl/share/locale\"" -DHAVE_CONFIG_H -I./../incl                                                                                                                                                             ude/opcode -I./../readline/.. -I./../zlib -I../bfd -I./../bfd -I./../include -I.                                                                                                                                                             ./libdecnumber -I./../libdecnumber  -I./gnulib/import -Ibuild-gnulib/import  -DT                                                                                                                                                             UI=1 -I/home/ubuntu/lede/staging_dir/host/include     -Wall -Wpointer-arith -Wno                                                                                                                                                             -unused -Wunused-value -Wunused-variable -Wunused-function -Wno-switch -Wno-char                                                                                                                                                             -subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -W                                                                                                                                                             no-sign-compare -Wno-error=maybe-uninitialized -Wsuggest-override -Wimplicit-fal                                                                                                                                                             lthrough=3 -Wduplicated-cond -Wshadow=local -Wformat -Wformat-nonliteral  -c -o                                                                                                                                                              disasm.o -MT disasm.o -MMD -MP -MF ./.deps/disasm.Tpo disasm.c
In file included from target.h:79:0,
                 from disasm.c:22:
btrace.h: In constructor 'btrace_function::btrace_function(minimal_symbol*, symb                                                                                                                                                             ol*, unsigned int, unsigned int, int)':
btrace.h:137:20: internal compiler error: Segmentation fault
       level (level_)
                    ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1637: recipe for target 'disasm.o' failed
make[6]: *** [disasm.o] Error 1
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5                                                                                                                                                             .0_musl/gdb-8.3.1/gdb'
Makefile:8792: recipe for target 'all-gdb' failed
make[5]: *** [all-gdb] Error 2
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5                                                                                                                                                             .0_musl/gdb-8.3.1'
Makefile:849: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5                                                                                                                                                             .0_musl/gdb-8.3.1'
Makefile:68: recipe for target '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc                                                                                                                                                             -7.5.0_musl/gdb-8.3.1/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gdb-8.                                                                                                                                                             3.1/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/toolchain/gdb'
time: toolchain/gdb/compile#141.06#37.29#199.20
toolchain/Makefile:98: recipe for target 'toolchain/gdb/compile' failed
make[2]: *** [toolchain/gdb/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
toolchain/Makefile:96: recipe for target '/home/ubuntu/lede/staging_dir/toolchai                                                                                                                                                             n-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stam                                                                                                                                                             p/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'

hplee0120, 2020-06-23:
> 请问下这是哪里出问题，还是按照之前的编译步骤，之前都是一次过的，这两天重装系统之后一直过不去，试过挺多办法了都没解决。ubuntu18和20都试了。make clean各种办法都不行
> 
> g++ -x c++ -g -O2 -I. -I. -I./config -DLOCALEDIR=""/home/ubuntu/lede/staging _dir/toolchain-x86_64_gcc-7.5.0_musl/share/locale"" -DHAVE_CONFIG_H -I./../incl ude/opcode -I./../readline/.. -I./../zlib -I../bfd -I./../bfd -I./../include -I. ./libdecnumber -I./../libdecnumber -I./gnulib/import -Ibuild-gnulib/import -DT UI=1 -I/home/ubuntu/lede/staging_dir/host/include -Wall -Wpointer-arith -Wno -unused -Wunused-value -Wunused-variable -Wunused-function -Wno-switch -Wno-char -subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -W no-sign-compare -Wno-error=maybe-uninitialized -Wsuggest-override -Wimplicit-fal lthrough=3 -Wduplicated-cond -Wshadow=local -Wformat -Wformat-nonliteral -c -o disasm.o -MT disasm.o -MMD -MP -MF ./.deps/disasm.Tpo disasm.c In file included from target.h:79:0, from disasm.c:22: btrace.h: In constructor 'btrace_function::btrace_function(minimal_symbol*, symb ol*, unsigned int, unsigned int, int)': btrace.h:137:20: internal compiler error: Segmentation fault level (level_)
> ^
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See file:///usr/share/doc/gcc-7/README.Bugs for instructions.
> Makefile:1637: recipe for target 'disasm.o' failed
> make[6]: *** [disasm.o] Error 1
> make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5 .0_musl/gdb-8.3.1/gdb'
> Makefile:8792: recipe for target 'all-gdb' failed
> make[5]: *** [all-gdb] Error 2
> make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5 .0_musl/gdb-8.3.1'
> Makefile:849: recipe for target 'all' failed
> make[4]: *** [all] Error 2
> make[4]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5 .0_musl/gdb-8.3.1'
> Makefile:68: recipe for target '/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc -7.5.0_musl/gdb-8.3.1/.built' failed
> make[3]: *** [/home/ubuntu/lede/build_dir/toolchain-x86_64_gcc-7.5.0_musl/gdb-8. 3.1/.built] Error 2
> make[3]: Leaving directory '/home/ubuntu/lede/toolchain/gdb'
> time: toolchain/gdb/compile#141.06#37.29#199.20
> toolchain/Makefile:98: recipe for target 'toolchain/gdb/compile' failed
> make[2]: *** [toolchain/gdb/compile] Error 2
> make[2]: Leaving directory '/home/ubuntu/lede'
> toolchain/Makefile:96: recipe for target '/home/ubuntu/lede/staging_dir/toolchai n-x86_64_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
> make[1]: *** [/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/stam p/.toolchain_compile] Error 2
> make[1]: Leaving directory '/home/ubuntu/lede'

NorwayLuo, 2020-06-23:
遇到同样的问题，18和20都试了。

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 * 	libubox20191228
 * opkg_install_cmd: Cannot install package base-files.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 * 	libubox20191228
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for urngd:
 * 	libubox20191228
 * opkg_install_cmd: Cannot install package urngd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/norway/lede'
package/Makefile:108: recipe for target '/home/norway/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/norway/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2

lqhorochi, 2020-06-23:
同样出错,,

9l, 2020-06-24:
网络问题没有下载全吧

Jose7871, 2020-06-24:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 *      libubox20191228
 * opkg_install_cmd: Cannot install package base-files.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for urngd:
 *      libubox20191228
 * opkg_install_cmd: Cannot install package urngd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/krystal/lede'
package/Makefile:108: recipe for target '/home/krystal/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/krystal/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/krystal/lede'
/home/krystal/lede/include/toplevel.mk:225: recipe for target 'world' failed

Me too, 1804 ltsc win10子系统

lqhorochi, 2020-06-25:
一样编译出错,....
貌似是GC10更新后就编译出错了

closed 2020-06-25 by hplee0120