coolsnowwolf/lede #4871: 378ca0f版本x86-legacy编译失败

soundghost, 2020-06-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
今天早上git pull到了最新的 378ca0f版本后x86-legacy（i586）版本就编译通不过了。
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
今天早上git pull到了最新的 378ca0f版本后x86-legacy（i586）版本就编译通不过了。

AmadeusGhost, 2020-06-24:
编译不过那报错日志呢？

soundghost, 2020-07-01:
好吧，老实说之前我一直不用单核心编译，因为lede的文件夹我一直都有备份（成功编译的备份），所以为了节省时间我一直都是全部核心在编译。
但是更新到最新的版本后，目前只有x86的版本编译出错，别的mtk核心等等多核心编译都不会出错。
所以为了查看错误信息，我昨天专门用了单核心编译，目前报错日志如下，请参考：

grub-mkimage-efi -d /home/myname/lede/staging_dir/host/lib/grub/x86_64-efi -o /home/myname/lede/build_dir/target-i386_pentium_musl/linux-x86_legacy/grub2.efi/efi/boot/bootx64.efi -O x86_64-efi -p /efi/boot -c /home/myname/lede/build_dir/target-i386_pentium_musl/linux-x86_legacy/grub2.efi/efi/boot/grub.cfg boot chain configfile ext2 linux ls part_msdos reboot serial part_gpt part_msdos search fat exfat ext2 efi_gop efi_uga gfxterm
grub-mkimage-efi: error: cannot open `/home/myname/lede/staging_dir/host/lib/grub/x86_64-efi/moddep.lst': No such file or directory.
Makefile:350: recipe for target 'install-images' failed
make[5]: *** [install-images] Error 1
make[5]: Leaving directory '/home/myname/lede/target/linux/x86/image'
Makefile:30: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/myname/lede/target/linux/x86'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/myname/lede/target/linux'
time: target/linux/install#1536.46#167.10#1682.16
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/myname/lede'
target/Makefile:19: recipe for target '/home/myname/lede/staging_dir/target-i386_pentium_musl/stamp/.target_install' failed
make[1]: *** [/home/myname/lede/staging_dir/target-i386_pentium_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/myname/lede'
/home/myname/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

今天来提交错误日志的时候也看到有人说x64的核心也报错。
估计应该是coolsnowwolf近期对x86做的改动导致的。
我也没有排错的能力，只能提交issue上来。
麻烦大家了。

coolsnowwolf, 2020-07-02:
已解决

closed 2020-07-02 by coolsnowwolf