coolsnowwolf/lede #4873: 如何只编译packages插件，不编译固件

AP0rPi7uQ9, 2020-06-24:
如何编译lede/pacakges/lean中的插件呢？

我不想编译固件，每次为了插件编译固件要1个多小时太耗时间了，我想用什么插件编译什么插件。

求大神指点怎么编译插件，最好直接能编译出来路由器后台可以直接安装的格式。

goldkeyber112, 2020-06-24:
编译SDK，然后用SDK编译package

KumaTea, 2020-06-24:
昨天刚看到：

在 `lede` 目录下，先 `make menuconfig` 选好你的型号，

然后 **`make package/path/to/package/compile`** ，其中 `path/to/package` 是插件路径，

比如 `make package/lean/luci-app-vlmcsd/compile V=s`

编译完成的 `ipk` 和它的所有依赖在 `bin/packages/arch/` 下。

Cathgao, 2020-06-25:
时间差不了多少，强迫症不想看到固件的话编译菜单不勾选任何image

AP0rPi7uQ9, 2020-06-26:
@goldkeyber112 
我的是新路由3，mt7621的，
https://downloads.openwrt.org/releases/19.07.3/targets/
问题1：在这个页面找不到mt7621的SDK，求教这是为什么呢
问题2：像LUCI图形化的APP，我听说sdk不能编译是吗？SDK在make menuconfig中找不到luci的app，应该怎么办呢？

AP0rPi7uQ9, 2020-06-26:
我用的是openwrt 19.07.3的版本

closed 2020-07-02 by coolsnowwolf