coolsnowwolf/lede #4876: 经常出现DHCP租约丢失的情况，日志中有大量的DHCPDISCOVER和DHCPOFFER，有反馈相同问题的情况

servicehost, 2020-06-24:
局域网中的设备，无论是lan或者vlan中的各种设备，经常会出现失去DHCP租约的情况，DHCP设置的租约时长为12小时，但实际上经常过1小时该设备就会失去DHCP租约，失去租约的设备会短时断线，设备类型包括：摄像头、手机、玩客云..等等，造成手游会掉线。
我曾经尝试：关掉IPV6，修改DHCP租约为24小时，皆无效。

该问题有另外一个人和我是一模一样的情况：
https://github.com/coolsnowwolf/lede/issues/4729

路由器型号和固件版本：
N3710软路由
固件版本 | OpenWrt  R20.6.18 (Compiled by Lusty on Jun.18, 2020) /LuCI Master (git-20.141.34051-f750b5f)
内核版本 | 4.19.122

详细日志：
`Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8464]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8464]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8465]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8465]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8467]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8467]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8465]: DHCPREQUEST(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPREQUEST(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPACK(br-lan) 192.168.1.181 00:22:6d:4b:29:64 WKY-A-02-BLACK-3456
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8465]: DHCPACK(br-lan) 192.168.1.181 00:22:6d:4b:29:64 WKY-A-02-BLACK-3456
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8464]: DHCPREQUEST(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8464]: DHCPACK(br-lan) 192.168.1.181 00:22:6d:4b:29:64 WKY-A-02-BLACK-3456
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8467]: DHCPREQUEST(br-lan) 192.168.1.181 00:22:6d:4b:29:64
Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8467]: DHCPACK(br-lan) 192.168.1.181 00:22:6d:4b:29:64 WKY-A-02-BLACK-3456
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8465]: DHCPDISCOVER(br-lan) 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8465]: DHCPOFFER(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8467]: DHCPDISCOVER(br-lan) 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8467]: DHCPOFFER(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8464]: DHCPDISCOVER(br-lan) 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8466]: DHCPDISCOVER(br-lan) 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8464]: DHCPOFFER(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8466]: DHCPOFFER(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8466]: DHCPREQUEST(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8466]: DHCPACK(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2 WKY-B-01-BLACK-9294
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8465]: DHCPREQUEST(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8465]: DHCPACK(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2 WKY-B-01-BLACK-9294
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8464]: DHCPREQUEST(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8464]: DHCPACK(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2 WKY-B-01-BLACK-9294
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8467]: DHCPREQUEST(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2
Wed Jun 24 21:19:46 2020 daemon.info dnsmasq-dhcp[8467]: DHCPACK(br-lan) 192.168.1.183 00:22:6d:4b:2c:e2 WKY-B-01-BLACK-9294`

我不是很懂，但经过google发现如下情况和我遇到的一模一样：
https://github.com/lede-project/source/pull/1079
https://bugs.openwrt.org/index.php?do=details&task_id=673
不知对能否解决该问题有所帮助。

shu0802, 2020-06-24:
自己编译的固件也有出现这样的问题，中间就断网几秒钟，自己又恢复正常了

Se7enchang, 2020-06-25:
曾经提过这个DHCP的问题，至于结果么，现在已刷回老固件2月份的没有此问题了，2月份后续的都会有类似的问题，如果不是实时的连网不会发现这种问题，最典型的连网6小时以上的在线游戏，6小时自动续期就会出现断几秒，重连连上，台式机网线直连玩游戏很直观的掉线。

servicehost, 2020-06-25:
> 
> 
> 曾经提过这个DHCP的问题，至于结果么，现在已刷回老固件2月份的没有此问题了，2月份后续的都会有类似的问题，如果不是实时的连网不会发现这种问题，最典型的连网6小时以上的在线游戏，6小时自动续期就会出现断几秒，重连连上，台式机网线直连玩游戏很直观的掉线。

请问你刷回的2月份固件版本及内核是什么？ 我也打算刷回去，这个突然闪断太耽误事了。

Se7enchang, 2020-06-25:
OpenWrt R20.2.15 / LuCI Master (git-20.046.55654-5343a8b)  内核 4.19.101，有条件先换回这个吧，家里有打游戏的才发现这个问题的，后边试过很多，就刷回老版本了。

servicehost, 2020-06-25:
> 
> 
> OpenWrt R20.2.15 / LuCI Master (git-20.046.55654-5343a8b) 内核 4.19.101，有条件先换回这个吧，家里有打游戏的才发现这个问题的，后边试过很多，就刷回老版本了。

谢谢

dolphinpaopao, 2020-06-25:
自编译x86固件，检查日志和日常使用，未发现相同问题。记得以前似乎有人提过可能是多播等和网络功能有关的组件造成的，我编译的时候是从来没有把docker和多播这些编译进去的，一是并不需要在路由系统里面实现这些功能，二是尽量减少可能的稳定性影响。

dolphinpaopao, 2020-06-25:
是否需要我自编译的x86 uefi固件来测试一下，这个固件网卡驱动只有r8168和i350的，组件就带了下图所示。
[image]

servicehost, 2020-06-25:
> 
> 
> 是否需要我自编译的x86 uefi固件来测试一下，这个固件网卡驱动只有r8168和i350的，组件就带了下图所示。
> [image]

谢谢，我的网卡是8111不一致，用的是combined-squashfs固件，不是UEFI的。

servicehost, 2020-06-25:
> 
> 
> 自己编译的固件也有出现这样的问题，中间就断网几秒钟，自己又恢复正常了

请问你自编译的固件也集成了多拨等网络组件了吗？楼下有人回复没有集成这些组件就未发生些问题。

shu0802, 2020-06-25:
> > 自己编译的固件也有出现这样的问题，中间就断网几秒钟，自己又恢复正常了
> 
> 请问你自编译的固件也集成了多拨等网络组件了吗？楼下有人回复没有集成这些组件就未发生些问题。
6月份之前编译的固件都没有任何问题，可能不是与插件有关

kongfl888, 2020-06-25:
同闪断，没开任何不可说等其他应用包括多拨负载等（因为没装，哈哈），就一个纯路由（开了samba36、minidlna、kms），也是最近开始出现，6月初是正常的。

servicehost, 2020-06-29:
经过楼上Se7enchang的指教，我也换回了：
固件版本 | OpenWrt  R20.2.15 / LuCI Master (git-20.045.63360-69eac90)
内核版本 | 4.19.101

实测发现丢失DHCP租约的情况大大减少（偶尔有但不频繁）
我自己没有编译固件的能力，所以使用的都是别人的固件。因此不能排除是不是固件编译者加入了某个插件或编译不当造成的。
这个问题应该就是dnsmasq-dhcp这个插件的问题，不知道编译时能否退选择该插件的老版本就能解决问题。

shu0802, 2020-06-29:
> **问题描述**
> 局域网中的设备，无论是lan或者vlan中的各种设备，经常会出现失去DHCP租约的情况，DHCP设置的租约时长为12小时，但实际上经常过1小时或更短时间该设备就会失去DHCP租约，失去租约的设备会短时断线，设备类型包括：摄像头、手机、玩客云..等等，造成手游会掉线。
> 我曾经尝试：关掉IPV6，修改DHCP租约为24小时，换全部网线为6类线，皆无效。
> 
> **该问题有另外一个人和我是一模一样的情况：**
> #4729
> 楼下回复的很多人也有同样的问题。
> 
> **路由器型号和固件版本：**
> N3710软路由
> 固件版本 | OpenWrt R20.6.18 (Compiled by Lusty on Jun.18, 2020) /LuCI Master (git-20.141.34051-f750b5f)
> 内核版本 | 4.19.122
> 固件来源（作者Lusty，使用lean源码编译）
> https://www.right.com.cn/forum/thread-3708606-1-1.html
> 
> **不是很懂，但经过google发现如下情况与此一致：**
> [lede-project/source#1079](https://github.com/lede-project/source/pull/1079)
> https://bugs.openwrt.org/index.php?do=details&task_id=673
> 不知对能否解决该问题有所帮助。
> 
> **详细日志（部份）：**
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8466]: DHCPDISCOVER(eth3.86) b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8467]: DHCPDISCOVER(eth3.86) b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8467]: DHCPOFFER(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8466]: DHCPOFFER(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8465]: DHCPDISCOVER(eth3.86) b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8465]: DHCPOFFER(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8464]: DHCPDISCOVER(eth3.86) b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8464]: DHCPOFFER(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8466]: DHCPREQUEST(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8467]: DHCPREQUEST(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8466]: DHCPACK(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15 ZQB-B-1-1T-Pro-Gold
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8467]: DHCPACK(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15 ZQB-B-1-1T-Pro-Gold
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8465]: DHCPREQUEST(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8465]: DHCPACK(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15 ZQB-B-1-1T-Pro-Gold
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8464]: DHCPREQUEST(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15
> Wed Jun 24 20:09:47 2020 daemon.info dnsmasq-dhcp[8464]: DHCPACK(eth3.86) 192.168.86.86 b0:d5:9d:5f:b8:15 ZQB-B-1-1T-Pro-Gold
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPDISCOVER(br-lan) b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPOFFER(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8466]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8465]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPREQUEST(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8464]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
> Wed Jun 24 20:56:51 2020 daemon.info dnsmasq-dhcp[8467]: DHCPACK(br-lan) 192.168.1.184 b0:d5:9d:d3:0a:e7 WKY-B-02-BLUE-0700
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8464]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8464]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8465]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8465]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8467]: DHCPDISCOVER(br-lan) 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8467]: DHCPOFFER(br-lan) 192.168.1.181 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8465]: DHCPREQUEST(br-lan) 192.168.1.181 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPREQUEST(br-lan) 192.168.1.181 00:22:6d:4b:29:64
> Wed Jun 24 20:57:07 2020 daemon.info dnsmasq-dhcp[8466]: DHCPACK(br-lan) 192.168.1.181 00:22:6d:4b:29:64 WKY-A-02-BLACK-3456
[... cut ...]
Tue Jun 30 00:59:55 2020 daemon.notice miniupnpd[16704]: Listening for NAT-PMP/PCP traffic on port 5351
Tue Jun 30 00:59:55 2020 user.notice firewall: Reloading firewall due to ifupdate of vwan1 (pppoe-vwan1)
Tue Jun 30 00:59:55 2020 cron.info crond[16797]: crond (busybox 1.30.0) started, log level 5
Tue Jun 30 00:59:55 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun 30 00:59:55 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun 30 01:00:00 2020 cron.info crond[16797]: USER root pid 16991 cmd echo "" > /tmp/ssrplus.log

shu0802, 2020-06-29:
从日志上看，与上面的不一样
Mon Jun 29 23:46:48 2020 daemon.notice pppd[11989]: remote LL address fe80::ce1a:faff:fee8:0aa0
Mon Jun 29 23:46:48 2020 daemon.notice netifd: Network device 'pppoe-vwan1' link is up
Mon Jun 29 23:46:48 2020 daemon.notice netifd: Interface 'vwan1' is now up
Mon Jun 29 23:46:48 2020 daemon.notice netifd: Network alias 'pppoe-vwan1' link is up
Mon Jun 29 23:46:48 2020 daemon.notice netifd: Interface 'vwan1_6' is enabled
Mon Jun 29 23:46:48 2020 daemon.notice netifd: Interface 'vwan1_6' has link connectivity
Mon Jun 29 23:46:48 2020 daemon.notice netifd: Interface 'vwan1_6' is setting up now
Mon Jun 29 23:46:48 2020 daemon.notice netifd: Interface 'vwan1_6' is now up
Mon Jun 29 23:46:49 2020 user.notice mwan3[12507]: Execute ifup event on interface vwan1 (pppoe-vwan1)
Mon Jun 29 23:46:49 2020 user.notice mwan3[12507]: Starting tracker on interface vwan1 (pppoe-vwan1)
Mon Jun 29 23:46:49 2020 user.notice mwan3track[23200]: Stopping mwan3track for interface "vwan1"
Mon Jun 29 23:46:49 2020 daemon.err pppd[12308]: syncppp: sem_timewait time out
Mon Jun 29 23:46:49 2020 daemon.err pppd[12308]: syncppp sync fail
Mon Jun 29 23:46:50 2020 daemon.info pppd[11993]: Remote message: 22105000: Too many sessions for this account!
Mon Jun 29 23:46:50 2020 daemon.err pppd[11993]: PAP authentication failed
Mon Jun 29 23:46:50 2020 daemon.notice pppd[11993]: Connection terminated.
Mon Jun 29 23:46:50 2020 daemon.info pppd[11993]: Sent PADT
Mon Jun 29 23:46:50 2020 daemon.info pppd[11993]: Exit.
Mon Jun 29 23:46:50 2020 daemon.notice netifd: Interface 'vwan3' is now down
Mon Jun 29 23:46:50 2020 daemon.notice netifd: Interface 'vwan3' is setting up now
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - slhc
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - ppp_generic
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - pppox
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - pppoe
Mon Jun 29 23:46:50 2020 daemon.notice netifd: vwan3 (12772): 1 processes have arrived, waiting for the left 4
Mon Jun 29 23:46:50 2020 user.info mwan3[12507]: connection tracking not flushed on interface vwan1 (pppoe-vwan1) ifup
Mon Jun 29 23:46:50 2020 user.notice firewall: Reloading firewall due to ifup of vwan1 (pppoe-vwan1)
Mon Jun 29 23:46:50 2020 cron.info crond[12945]: crond (busybox 1.30.0) started, log level 5
Mon Jun 29 23:46:50 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jun 29 23:46:50 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jun 29 23:46:50 2020 daemon.info pppd[12308]: Remote message: 22105000: Too many sessions for this account!
Mon Jun 29 23:46:50 2020 daemon.err pppd[12308]: PAP authentication failed
Mon Jun 29 23:46:50 2020 daemon.notice pppd[12308]: Connection terminated.
Mon Jun 29 23:46:50 2020 daemon.info pppd[12308]: Sent PADT
Mon Jun 29 23:46:50 2020 daemon.info pppd[12308]: Exit.
Mon Jun 29 23:46:50 2020 daemon.notice netifd: Interface 'vwan2' is now down
Mon Jun 29 23:46:50 2020 daemon.notice netifd: Interface 'vwan2' is setting up now
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - slhc
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - ppp_generic
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - pppox
Mon Jun 29 23:46:50 2020 daemon.err insmod: module is already loaded - pppoe

xpc0938, 2020-06-29:
> 从日志上看，与上面的不一样
> Mon Jun 29 23:46:48 2020 daemon.notice pppd[11989]: remote LL address fe80::ce1a:faff:fee8:0aa0
> Mon Jun 29 23:46:48 2020 daemon.notice netifd: Network device 'pppoe-vwan1' link is up [...]

你看日志:pppd[12308]: Remote message: 22105000: Too many sessions for this account!
感觉是你的宽带不支持多拨，或者你设置的多拨路数太多了，如果不支持就可以卸载mwan功能。

zerofo, 2020-07-01:
我昨天 试过整个dhcp死掉...重启无法解决... 好像是配置文件都不见了。。

shu0802, 2020-07-02:
> 你看日志:pppd[12308]: Remote message: 22105000: Too many sessions for this account!
> 感觉是你的宽带不支持多拨，或者你设置的多拨路数太多了，如果不支持就可以卸载mwan功能。

我的宽带支持虚拟多拨和物理多拨的

closed 2020-07-02 by coolsnowwolf